esphome/feature-requests #758: Wifi channel as exposed sensor

dvdavide, 2020-06-06:
**Describe the problem you have/What new integration you would like**

I'd like to be able to have the wifi channel that the device is using available as a sensor to use with HA
**Please describe your use case for this integration and alternatives you've tried:**

I have a few access points with the same SSID on different channels, and I would like to check at a glance that the devices are connected to the right AP
**Additional context**

If the sensor was available I could group the devices by channel and I could see what AP they are connected to, and also make them perform a reboot if they are connected to a clearly "wrong" AP (ie not the closest one)

Wauter, 2020-06-08:
Hi @dvdavide 

This is not currently part of esphome, but seems fairly trivial to achieve with a template (text)  sensor. I verified this on esp6288

`text_sensor:
  - platform: template
    name: "Wifi BSSID"
    lambda: |-
      std::string out;
      char buffer[64];
      sprintf(buffer, "%u", wifi_get_channel());
      out.append(buffer);
      return out;
    update_interval: 60s`

output:
[14:22:00][D][text_sensor:015]: 'Wifi BSSID': Sending state '11'

Wauter, 2020-06-08:
( Also, if you have a clear definition of which access point you would like each sensor to be connected to, you can define that access point's bssid in your configuration: https://esphome.io/components/wifi.html?highlight=bssid )

dvdavide, 2020-06-08:
All excellent advice, thank you very much

Wauter, 2020-06-19:
I tested it on an esp32 device before posting. I am running the dev build of esphome though, so if you are not, you could try again on dev?

sergeolkhovik, 2020-06-23:
Just tested with latest dev build via docker and still the same error:

```
Compiling .pioenvs/d4/lib789/WiFi/ETH.cpp.o
src/main.cpp: In lambda function:
src/main.cpp:260:46: error: 'wifi_get_channel' was not declared in this scope
       sprintf(buffer, "%u", wifi_get_channel());
                                              ^
Compiling .pioenvs/d4/lib789/WiFi/WiFi.cpp.o
*** [.pioenvs/d4/src/main.cpp.o] Error 1
```

I don't know is that important or not, I'm testing on esp32-cam module, and here's my full config:

```
esphome:
  name: d4
  platform: ESP32
  #board: esp32cam
  board: esp32dev
  #platformio_options:
  #  board_build.partitions: min_spiffs.csv
  on_boot:
      then:
        light.turn_off: builtin_led

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

# Enable logging
logger:

# Enable Home Assistant API
api:
    password: !secret api_password

ota:
    password: !secret ota_password
sensor:
  # Sensors with general information.
  # Uptime sensor.
  - platform: uptime
    name: D4 Uptime
    # WiFi Signal sensor.
  - platform: wifi_signal
    name: D4 WiFi Signal
    update_interval: 60s

output:
  - platform: gpio
    pin: GPIO4
    id: builtin_led_output

light:
  - platform: binary
    id: builtin_led
    name: D4 Built-in LED
    output: builtin_led_output

text_sensor:
  # Text sensors with general information.
  # Expose ESPHome version as sensor.
  - platform: version
    name: D4 ESPHome Version
    # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: D4 IP address
  - platform: wifi_info
    bssid:
      name: D4 BSSID
  - platform: template
    name: D4 Channel
    lambda: |-
      std::string out;
      char buffer[64];
      sprintf(buffer, "%u", wifi_get_channel());
      out.append(buffer);
      return out;
    update_interval: 60s

binary_sensor:
  - platform: status
    name: D4 Status
  - platform: gpio
    pin: GPIO2
    name: "Entrance PIR Sensor"
    device_class: motion

switch:
  # Switch to restart the doorbell.
  - platform: restart
    name: D4 Restart

## Example configuration entry
#esp32_camera:
#  external_clock:
#    pin: GPIO0
#    frequency: 20MHz
#  i2c_pins:
#    sda: GPIO26
#    scl: GPIO27
#  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
#  vsync_pin: GPIO25
#  href_pin: GPIO23
#  pixel_clock_pin: GPIO22
#  power_down_pin: GPIO32
#  # Image settings
#  name: Entrance Camera
#  jpeg_quality: 25
#  #resolution: 640x480
#  resolution: 800x600
#  #resolution: 1024x768
#  #resolution: SXGA
#  #resolution: UXGA
#  vertical_flip: false
#  horizontal_mirror: false
```

Also may test on pure esp32 module tomorrow.

PS: I commented out camera definition, want to test if camera module heats esp32 and thus affects on PIR sensor sensitivity.

schinckel, 2020-06-24:
FWIW, I wanted the same as this (but the BSSID):

    text_sensor:
      - platform: template
        name: "BSSID"
        lambda: |-
          std::string out;
          char buffer[64];
          uint8_t b[6] = {};
          uint8_t *raw_bssid = WiFi.BSSID();
          if (raw_bssid != nullptr)
            memcpy(b, raw_bssid, sizeof(b));
          sprintf(buffer, "%02X:%02X:%02X:%02X:%02X:%02X", b[0], b[1], b[2], b[3], b[4], b[5]);
          out.append(buffer);
          return out;
        update_interval: 60s

sergeolkhovik, 2020-06-24:
Well, BSSID is already available:

```
  - platform: wifi_info
    bssid:
      name: D4 BSSID
```

Works well both on ESP8266 and ESP32.

schinckel, 2020-06-24:
Oh, thanks. I missed this whole sensor in the docs!

sergeolkhovik, 2020-07-02:
> I tested it on an esp32 device before posting. I am running the dev build of esphome though, so if you are not, you could try again on dev?

I've got this ESP32 board (https://www.aliexpress.com/item/32799253567.html), and I specify "esp32doit-devkit-v1" as board in my yaml file. And I still get the same error, unfortunately :( I tested both with dev version I did previously and the latest.

sergeolkhovik, 2020-07-02:
BTW what PIO version to do you have in your docker image?
I see this in the output:
```
Obsolete PIO Core v4.3.3 is used (previous was 4.3.4)
```

Since this is a docker image, I expect PIO will be the latest?

sergeolkhovik, 2020-07-02:
Gentlemen, I just found this idea: https://forum.mongoose-os.com/discussion/1529/wi-fi-connection-rssi-signal-strength
So here's my implementation:
```
  - platform: template
    name: D13 Channel
    lambda: |-
      wifi_ap_record_t wifidata;
      esp_wifi_sta_get_ap_info(&wifidata);
      std::string out;
      if (wifidata.primary != 0) {
        char buffer[64];
        sprintf(buffer, "%u", wifidata.primary);
        out.append(buffer);
      }
      return out;
    update_interval: 60s
```

And it works:

[image]

TotallyInformation, 2021-04-03:
I get the same error as @sergeolkhovik 

```
src/main.cpp: In lambda function:
src/main.cpp:990:46: error: 'wifi_get_channel' was not declared in this scope
       sprintf(buffer, "%u", wifi_get_channel());
                                              ^
Compiling .pioenvs/ttgo01/libfa7/AsyncMqttClient-esphome/AsyncMqttClient/Packets/ConnAckPacket.cpp.o
*** [.pioenvs/ttgo01/src/main.cpp.o] Error 1
```

I'm using the latest build of esphome with a TTGO T-Display (using board `featheresp` as shown in the example for the `st7789v` display).

```
Processing ttgo01 (board: ttgo-t1; framework: arduino; platform: espressif32@1.12.4)
-----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif32 3.10004.210126 (1.0.4)
 - tool-esptoolpy 1.20600.0 (2.6.0)
 - toolchain-xtensa32 2.50200.97 (5.2.0)
Library Manager: Installing Hash
Library Manager: Already installed, built-in library
Dependency Graph
|-- <AsyncTCP-esphome> 1.1.1
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <FS> 1.0
|-- <ESPAsyncWebServer-esphome> 1.2.7
|   |-- <AsyncTCP-esphome> 1.1.1
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|   |-- <ArduinoJson-esphomelib> 5.13.3
|-- <AsyncMqttClient-esphome> 0.8.4
|   |-- <AsyncTCP-esphome> 1.1.1
|-- <ArduinoJson-esphomelib> 5.13.3
|-- <Update> 1.0
|-- <WiFi> 1.0
|-- <SPI> 1.0
```

The implementation from the last post above worked for me.

slimcdk, 2021-06-24:
This is working fine for me.

```yaml
sensor:
  - platform: template
    name: Wifi Channel
    id: wifi_channel
    lambda: return WiFi.channel();
    update_interval: 60s
```

closed 2022-07-01 by nagyrobi