esphome/feature-requests #760: Add support for Adafruit SSD1331 OLED color display

hapklaar, 2020-06-07:
**Describe the problem you have/What new integration you would like**
Please add support for the cheap Adafruit SSD1331 96x64 OLED color display.

**Please describe your use case for this integration and alternatives you've tried:**
Currently unable to use this display with ESPHome as it doesn't seem to be supported. Also wasn't able to find a similar platform (is there a list of supported platforms anywhere?)

**Additional context**
This is a simple SPI controlled display. Library is available from Adafruit (Adafruit_SSD1331.h)

kbx81, 2020-06-16:
I've recently picked up a couple of their SSD1351 displays -- hopefully the SSD1331 is similar enough that it can be part of the same component. I may take a stab at implementing this in the not-too-distant future, unless someone beats me to it. 😄

kbx81, 2020-06-25:
Unfortunately this display isn't quite as similar to the SSD1351 as it probably should be to roll into a single component. Next time I order some hardware, I just might add one of these to the order so I can properly build and test it.

hapklaar, 2020-07-28:
I see you finished the SSD1351 integration, great! Let me know if it would help if I send you one of these SSD1331's.

kbx81, 2020-07-28:
I have one of these on order, but it seems everybody is out of stock and they are not due to ship until late October. 😞

If you would like this added sooner and are willing to send one, I won't turn it down! 😄

hapklaar, 2020-07-28:
October really? You're not ordering from Ali I guess :)

If you PM me your address, I'll see to that you get one delivered!

kbx81, 2020-08-08:
@hapklaar I received the display -- thank you! 🎉
It's been implemented and PR'd. Just needs approval and merging. 😄

hapklaar, 2020-08-10:
@kbx81 That's great news, you're fast!

Will have a go at it as soon as it's available. Thanks again! :)

emeyedeejay, 2020-12-15:
Hi! Is there any update on this at all? Also keen to get my SSD1331 screen integrated into ESPHome.

Thanks!

kbx81, 2020-12-15:
It was merged into `dev` a while ago. It'll be in the next release, or if you need it now, you can run `dev`. 😄

emeyedeejay, 2020-12-16:
Thanks!

I found the updated documents and am now running dev. Still can't get it working though :D

emeyedeejay, 2020-12-16:
Now working 🙌

[image]

closed 2021-12-22 by hapklaar