esphome/feature-requests #762: Serial USB port on WEMO D1

AquaMCU, 2020-06-08:
**Describe the problem you have/What new integration you would like**

I want to use the USB serial communication so I can send commands over this port. For example my usecase:

I want to enable a controllino (a simple arduino device with a USB port with wifi using esphome

**Please describe your use case for this integration and alternatives you've tried:**

N/A
**Additional context**

Since most 8266 devices have a USB serial interface for flashing, I think it is a good idea if we could use it. It opens up tons of usecases.

I wonder if the UART device can already be used, but I did not find any tutorial on that.

randybb, 2020-06-08:
On these DEV boards is just a simple USB to serial converter attached to UART0 as you would do with a bare ESP8266/ESP32 module. Sure, you can use it, just need to disable logging as is described here https://esphome.io/components/uart.html

Probably you have noticed, that the name ESPHome stands for remote sensors/actuators running on ESP controlled from Home Assistant, so the primary way how you should control it is via HA API to have it everything super simple + mqtt and REST API are supported for these who like to make their lives harder.

I am not sure how interrupts on uart are handled... but you can take a look on [tuya](https://esphome.io/components/light/tuya.html) or [Custom UART Device](https://esphome.io/custom/uart.html)...

AquaMCU, 2020-06-08:
Hi.

I will give UART0 a try and come back to you.

Thanks for the reply

glmnet, 2020-07-02:
Yes, USB on wemos, etc, are a a fixed USB to Serial, USB cannot be used for anything else. It is connected to `UART0` by design.

closed 2020-07-02 by glmnet