esphome/feature-requests #764: Add "before" turn_on/turn_off trigger

JohanBraeken, 2020-06-08:
**Describe the problem you have/What new integration you would like**

In order to ensure no 2 outputs can be active at the same time, I'd like to have to possibility to execute local code "before" an output is changed state.
The current triggers "on_turn_on", "on_turn_off" execute code "after" the output has been changed and this is not always desireable.

**Please describe your use case for this integration and alternatives you've tried:**

A use case for this would be the automation of sunshades / covers.

One output triggers the "up" signal, another output the "down" signal.
Depending on the make and model, care must be taken to only activate a single output at any time, and most of the time a short delay must be present before changing directions.
Currently this can not be implemented in local code, but with the addition of a "before" trigger it will be:
The "before" trigger for turning on output 1, will execute code to deactivate output 2 and a small delay. Only after this code is completely executed, output 1 will be turned on.

In general, it would allow all integrations requiring only a single output to be active at the same time, such as automating something with a selection switch.

Additionaly, This can be used to ensure a certain state is set, before any additional action is taken.

**Additional context**

There might be a risk of loops if part of the actions in the "before" script includes the object that contains this trigger.

Wauter, 2020-06-08:
Have a look at gpio interlocking:
https://esphome.io/components/switch/gpio.html?highlight=interlock#interlocking

 and the template cover component. 
https://esphome.io/components/cover/template.html

JohanBraeken, 2020-06-08:
Thank you for your swift response.
Indeed, the referenced documentation completely implements the feature I described.

This request can be closed then.

closed 2020-06-08 by JohanBraeken