esphome/feature-requests #766: Atlas Scientific EZO i2c sensors

Neukam, 2020-06-08:
**Describe the problem you have/What new integration you would like**

I would like to see support for Atlas Scientific EZO sensors with i2c interface, specifically the ORP and PH sensors

**Please describe your use case for this integration and alternatives you've tried:**

 I want to be able to fully automate my pool with salt water generator. I'm currently using a Sonoff 4ch running Tasmota to control the pump, SWG and lights and read from a DS18B20 temperature sensor. I would like to be able to add the Atlas Scientific sensors and switch to ESPHome so that i can control the runtime of the SWG based on ORP, and also monitor PH.  These sensor modules could also be useful for a hydroponics system.

**Additional context**

There is a [library with examples](https://github.com/Atlas-Scientific/Ezo_I2c_lib).
There are datasheets and documentation on the product pages.
The following EZO devices are available
[ORP](https://www.atlas-scientific.com/circuits/ezo-orp-circuit/)
[PH](https://www.atlas-scientific.com/circuits/ezo-ph-circuit/)
[Dissolved Oxygen](https://www.atlas-scientific.com/circuits/ezo-dissolved-oxygen-circuit/)
[Electrical Conductivity](https://www.atlas-scientific.com/circuits/ezo-conductivity-circuit/)
[RTD Temperature](https://www.atlas-scientific.com/circuits/ezo-rtd-temperature-circuit/)
[Flow Meter Totalizer](https://www.atlas-scientific.com/circuits/ezo-embedded-flow-meter-totalizer/) This could be useful for tracking domestic water usage
[50 psi Pressure Sensor](https://www.atlas-scientific.com/probes/ezo-prs-embedded-pressure-sensor/)
[Peristaltic Dosing Pump](https://www.atlas-scientific.com/peristaltic/ezo-pmp-embedded-dosing-pump/)
[RGB Color Sensor](https://www.atlas-scientific.com/probes/ezo-rgb-embedded-color-sensor/)

Atlas Scientific also has a [Hydroponics Kit](https://www.atlas-scientific.com/kits/wi-fi-hydroponics-kit/) which has an ESP8266 mounted on a carrier board with PH, Electrical Conductivity, and RTD Temperature EZO boards with a provision to add one additional EZO board.  This could also be very useful for a pool controller.  Their code for the ESP is open source and [posted on their github](https://github.com/Atlas-Scientific/Ezo_I2c_lib/blob/master/Examples/hydroponics_kit/hydroponics_kit.ino).

mrgoose2, 2020-07-10:
User ssieb has a custom component for these sensors. github com/ssieb/custom_components/tree/master/ezo

I have my sensors on a tentacle shield attached to an arduino mega, the scl and sda pins from the mega are attached to the D2 and D1 pins on a NodeMCU. 

I created a folder called "custom-components" under my esphome folder and placed ssieb's "ezo" folder in it. 

Here is the relevant part of my yaml file.
`i2c:
  sda: D2
  scl: D1
  scan: True
  id: bus_a

  #ezo platform from: https://github.com/ssieb/custom_components
sensor:
  - platform: ezo
    id: ph_ezo
    name: "pH"
    address: 99
    accuracy_decimals: 2
    unit_of_measurement: "pH"

      
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 101
    accuracy_decimals: 2
    unit_of_measurement: "°C"

      
  - platform: ezo
    id: ec_ezo
    name: "EC CN"
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "CN"
`

I don't have much experience with esphome or arduino, but ssied's component made this easy.

ssieb, 2020-07-10:
Ok, fine, I'll write up the docs and file a PR. :-)

mrgoose2, 2020-07-10:
> Ok, fine, I'll write up the docs and file a PR. :-)

I was trying to get these sensors to work for the last couple days, stumbled across your repo and presto I was up and running. Thanks for putting it out there, I really appreciate it! So much that I had to share on this thread!

Neukam, 2020-07-11:
Sweet!  You guys are awesome!  I'll be trying this out this weekend

iHakia, 2020-07-22:
Hi guys, Thanks for the help implementing those sensors. 
I'm really trying to get this one working. for now, it seems that I have problem with the esphome folder structure or something. 
the files are located at \config(where my HA configuration yaml)\esphome\custom_components\ezo
this is my code 
```
esphome:
  name: nfatxxxxxlas
  platform: ESP8266
  board: esp01_1m
  includes:
    - ezo.h

i2c:
  sda: 4
  scl: 5
  scan: True
  id: bus_a

wifi:
  ssid: "xxxxxx"
  password: "xxxxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "xxxxFallback Hotspot"
    password: "LYoPZoHB4pdh"
    
sensor:
  - platform: ezo
    id: ph_ezo
    address: 99
    unit_of_measurement: "pH"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "xxxxxxxx"

ota:
  password: "xxxxxxxxxx"

```

Unfortunately I keep getting the message 
"platform not found. 'sensor. ezo'"
anyone have an idea? 
Help much appreciated 👍

ssieb, 2020-07-22:
Don't use the include.  Do you have all the files, including the two python ones in the ezo directory?

iHakia, 2020-07-23:
> Don't use the include. Do you have all the files, including the two python ones in the ezo directory?
I guess that yes. 
I removed the include. still not working :/
[image]

ssieb, 2020-07-23:
Which esphome version are you using?  You might need to use dev.

mrgoose2, 2020-07-24:
Double check your folder path. Here is mine. It looks like I had a typo in my instructions above. 
[image]

ssieb, 2020-07-24:
That looks the same as his.  It's likely that it needs something that is only in dev.  It would be something in the config parsing because that would make it not find the sensor code.  If you use the cli esphome to compile it, you would see the error message.

mrgoose2, 2020-07-24:
They typed it with the underscore but the picture of their config has a hyphen. Unless I'm not seeing something since I'm on mobile.

ssieb, 2020-07-25:
Oh, good catch!  I completely missed that. :facepalm:

iHakia, 2020-07-27:
> Oh, good catch! I completely missed that. 🤦

thanks mrgoose2 - That Worked!!!
Thanks for that 👍

iHakia, 2020-07-27:
So now we now that custom_componenets works while using normal esphome version. 
Now I have all my EZP sensors working and I this is how it's look on my config side. 
 Is there a way to do the Calibration side of things also? 
PH calibration
EC calibration 
DO calibration
?

`sensor:
  - platform: ezo
    id: ph_ezo
    address: 99
    unit_of_measurement: "pH"
    update_interval: 10s

  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    update_interval: 10s
    
  - platform: ezo
    id: ec_ezo
    name: "EC"
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "PPM"
    update_interval: 10s
    
  - platform: ezo
    id: do_ezo
    name: "DO"
    address: 97
    accuracy_decimals: 0
    unit_of_measurement: "mg/L"
    update_interval: 10s`

ssieb, 2020-07-27:
You can do the temperature calibration in a lambda using `id(ph_ezo).set_tempcomp_value(rtd_ezo.state);` for example.  You need to decide when you want to calibrate it based on time or change in temperature.

ssieb, 2020-07-27:
I haven't quite finished the documentation for the PR yet.

iHakia, 2020-07-27:
> You can do the temperature calibration in a lambda using `id(ph_ezo).set_tempcomp_value(rtd_ezo.state);` for example. You need to decide when you want to calibrate it based on time or change in temperature.

You are great, How about the pH calibration? 
In the Atlas documents the commands are 
ph:cal,mid,7     calibrate to pH 7                                         
ph:cal,low,4     calibrate to pH 4                                         
ph:cal,high,10   calibrate to pH 10                                        
ph:cal,clear     clear calibration    

is this can be done?

ssieb, 2020-07-27:
I'll have to look up the docs again.  I don't have any of these sensors and I had only been asked about the temperature compensation.

ssieb, 2020-08-03:
I checked the docs and the calibration really should be done from a PC.  It's not something that you're going to do regularly, it's mostly a one-time action.
I've now made a pull request for bringing the component into esphome.

iHakia, 2020-08-03:
> I checked the docs and the calibration really should be done from a PC. It's not something that you're going to do regularly, it's mostly a one-time action.
> I've now made a pull request for bringing the component into esphome.

Well I can say that until now it works great. 
The calibration part of the pH is highly important and should be done every month (for hydroponic use) and in order to do it from pc you basically have to reflash the esp with atlas firmware and use their uart command line. it's a harsh task to do. 

is there a way to send i2c commands to the module?

ssieb, 2020-08-03:
I don't think Atlas has ESP firmware.  Their firmware is in the sensor board.  You connect the sensor board to the PC to send it commands.  Nothing to do with the ESP device.  How have you been doing the calibrations now?  And how did you get it into i2c mode?
You could send arbitrary i2c commands using lambdas if you really wanted to.

iHakia, 2020-08-03:
Check this out 
https://www.atlas-scientific.com/files/Wi-Fi-Pool-kit-setup-guide.pdf
it's based on esp. 
maybe we can fish it out somehow. 
also, I don't really know how to use lambdas, 
Any good read recommendation?

ssieb, 2020-08-03:
Unless you're intending to make that, it's much more complicated than you need.  Just follow the instructions in https://www.atlas-scientific.com/files/pH_EZO_Datasheet.pdf.  It's too complicated to add the calibrations to esphome and out of scope.
You can read about lambdas at https://esphome.io/guides/automations.html.  It's a method of embedding C++ code in your esphome configs.

krazykatz69, 2020-11-05:
Hi ssieb,

I bought the Wi-Fi Hydroponics Kit, do you know if it is possible to get this to work over i2c?
I am getting an error saying Found no i2c devices.
See below:

[21:32:31][C][i2c:028]: I2C Bus:
[21:32:31][C][i2c:029]:   SDA Pin: GPIO4
[21:32:31][C][i2c:030]:   SCL Pin: GPIO5
[21:32:31][C][i2c:031]:   Frequency: 50000 Hz
[21:32:31][I][i2c:033]: Scanning i2c bus for active devices...
[21:32:31][I][i2c:049]: Found no i2c devices!
[21:32:31][C][logger:185]: Logger:
[21:32:31][C][logger:186]:   Level: DEBUG
[21:32:31][C][logger:187]:   Log Baud Rate: 115200
[21:32:31][C][logger:188]:   Hardware UART: UART0
[21:32:31][C][ezo.sensor:014]: EZO 'ph_ezo'
[21:32:31][C][ezo.sensor:014]:   Unit of Measurement: 'pH'
[21:32:31][C][ezo.sensor:014]:   Accuracy Decimals: 2
[21:32:31][C][ezo.sensor:015]:   Address: 0x63
[21:32:31][C][ezo.sensor:018]:   Update Interval: 60.0s
[21:32:31][C][ezo.sensor:014]: EZO 'RTD Temperature'
[21:32:31][C][ezo.sensor:014]:   Unit of Measurement: '°C'
[21:32:31][C][ezo.sensor:014]:   Accuracy Decimals: 2
[21:32:31][C][ezo.sensor:015]:   Address: 0x66
[21:32:31][C][ezo.sensor:018]:   Update Interval: 60.0s
[21:32:31][C][ezo.sensor:014]: EZO 'EC CN'
[21:32:31][C][ezo.sensor:014]:   Unit of Measurement: 'CN'
[21:32:31][C][ezo.sensor:014]:   Accuracy Decimals: 0
[21:32:31][C][ezo.sensor:015]:   Address: 0x64
[21:32:31][C][ezo.sensor:018]:   Update Interval: 60.0s
[21:32:31][C][captive_portal:169]: Captive Portal:
[21:32:31][C][ota:029]: Over-The-Air Updates:
[21:32:31][C][ota:030]:   Address: 192.168.0.144:8266
[21:32:31][C][ota:032]:   Using Password.
[21:32:31][C][api:095]: API Server:
[21:32:31][C][api:096]:   Address: 192.168.0.144:6053
[21:32:32][W][i2c:086]: Requesting 20 bytes from 0x64 failed!
[21:32:32][E][ezo.sensor:049]: read error
[21:32:34][W][i2c:070]: Received NACK on transmit of address 0x66
[21:32:34][W][i2c:086]: Requesting 20 bytes from 0x66 failed!

krazykatz69, 2020-11-05:
I think that I may just need to enable pin 14, 12 and 15 on the huzzah esp8266 to power the EZO circuit board , but am unsure how to achieve this.

krazykatz69, 2020-11-05:
Got it working, added the below to the code:

switch:
  - platform: gpio
    pin: 14
    name: "pH On"
    inverted: no
    
  - platform: gpio
    pin: 15
    name: "RTD Temp On"
    inverted: yes

  - platform: gpio
    pin: 12
    name: "EC On"
    inverted: no

krazykatz69, 2020-11-06:
I can see that the sensors are working in the ESPhome logs but don't know how to add them as entities in HA.
I have enable newly discovered entities enabled in esphome.

I have another esp8266 which is up and running and can see all 5 entities connected to that one.

Any help would be much appreciated.

ssieb, 2020-11-06:
This isn't the right place for getting help for things like that.  The esphome Discord channel would be best or the home assistant forum.

closed 2020-11-17 by jesserockz

oleokoong, 2021-03-14:
@krazykatz69 hi do you mind to share ur code? i follow urs instruction for wifi hydroponic still fail to get reading.

[22:42:31][W][i2c:076]: Unknown transmit error 4 for address 0x64
[22:42:31][W][i2c:076]: Unknown transmit error 4 for address 0x66
[22:42:31][W][i2c:076]: Unknown transmit error 4 for address 0x63

Morbidbystandard, 2021-03-28:
Tasmota now has calibration under i2c operation. Some of us like to calibrate weekly during a reservoir change.

jesserockz, 2021-03-28:
> Tasmota now has calibration under i2c operation. Some of us like to calibrate weekly during a reservoir change.

Please create a new feature-request for this.

iyotee, 2021-06-14:
> id(ph_ezo).set_tempcomp_value(rtd_ezo.state);

Can you please give an exemple, i always have a : Lambas is an invalid option for sensor.ezo. Please check the indentation.

J'ai tenté toute les indentation possible. Merci 

switch:
  - platform: gpio
    name: "Generic Output"
    pin : 12
    inverted: no
  - platform: gpio
    name: "Generic Output 2"
    pin: 14
    inverted: no
  - platform: gpio
    name: "Generic Output 3"
    pin : 15
    inverted: yes

    
sensor:
  - platform: ezo
    id: ec_ezo
    name: "ec"
    address: 100
    unit_of_measurement: "uS/cm"
    update_interval: 2s
    accuracy_decimals : 3
  - platform: ezo
    id: ph_ezo
    name: "ph"
    address: 99
    unit_of_measurement: "pH"
    update_interval: 2s
    accuracy_decimals : 3
    lambda: !lambda |-
       id(ph_ezo).set_tempcomp_value(rtd_ezo.state);
  - platform: ezo
    id: rtd_ezo

nilsi, 2021-09-26:
Hi @ssieb did you ever manage to get the PR out there for this? Very interested as well. Will probably try your custom component if not.

ssieb, 2021-09-26:
https://github.com/esphome/esphome/pull/1239 was merged almost a year ago.  See https://esphome.io/components/sensor/ezo.html