esphome/feature-requests #769: EspHome Native API: explicit HA IP address

mspinolo, 2020-06-09:
**Disclaimer:: I am not a network expert hence my statements could be wrong

Hi,

logging my network activity with Wireshark I see Esphome flashed ESP8266/ESP32 send a decent amount of broadcasting messages querying for home assistant IP address.

This leads to a load of the network and IoT devices has usually a limited wifi bandwidth: a huge number of devices could reduce it.

I was wondering if it could be possible (and meaningful) to add to native API component an parameter to make explicit which is HA instance IP address in order to avoid broadcasting.**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

mspinolo, 2020-06-11:
I was wondering if similar result could be obtained disabling mDNS on ESPHome devices like it was (I think) before 1.11.0 --> is there an option for this?

mddeff, 2020-06-21:
Would love to be able to manually specify a HA server in the config (either Hostname or IP).  Honestly I'm surprised ESPHome *requires* mdns/same subnet configurations.  (If I'm missing something, please feel free to enlighten)

glmnet, 2020-06-24:
It’s not simple. 

Esphome is actually a server. I.e. hass connects to ESPHome. 

There is an old fr about reverting this, which solves other issues too.

mspinolo, 2020-06-24:
Hi, 

So being ESPHOME a server why it is sending mDNS packets around?

glmnet, 2020-07-25:
Will be part of https://github.com/esphome/feature-requests/issues/46

closed 2020-07-25 by glmnet