esphome/feature-requests #770: Add support for BL0940 power metering IC

ClausRN, 2020-06-11:
**Describe the problem you have/What new integration you would like**

I would like support for this power metering chip
http://www.belling.com.cn/media/file_object/bel_product/BL0940/datasheet/BL0940_V1.1_en.pdf

**Please describe your use case for this integration and alternatives you've tried:**

It is used in the BlitzWolf® BW-SHP10 smart plug
https://templates.blakadder.com/blitzwolf_SHP10P

**Additional context**

Proxx, 2020-06-12:
The guys of tasmota are also making progress on this IC
https://github.com/arendst/Tasmota/issues/8175

geerdkakes, 2020-08-29:
Has someone already started working on this? 

I could not order the older BW-SHP6 anymore. The BW-SHP10 I just received could be flashed with Tuya-Convert. Relay is on GPIO05.

klaasnicolaas, 2021-04-10:
I would like to take up this issue, even now a few of those SHP10 plugs and it is a shame that it does not work with ESPHome yet.

However, I am still a beginner so don't expect much from it 😅

closed 2022-01-04 by oxan