esphome/feature-requests #771: Add support for SparkFun MPL3115A2 Barometric Pressure sensor

matthew-larner, 2020-06-11:
**Describe the problem you have/What new integration you would like**
Add support for SparkFun MPL3115A2 Barometric Pressure sensor
Library: https://github.com/sparkfun/SparkFun_MPL3115A2_Breakout_Arduino_Library

**Please describe your use case for this integration and alternatives you've tried:**
I would like to use the MPL3115A2 with a WEMOS LOLIN D32

**Additional context**

M-Wahren, 2020-11-05:
I would also like this to be added

matthew-larner, 2020-11-19:
It used to be there. I'm not sure if it was removed on purpose or it's a bug.

Just for clarity, I'm referring to the complication that when tapped open the actions menu like this: https://community-assets.home-assistant.io/original/3X/4/a/4a305ac7d4c683601b9e65c03cda3a40d9aad560.png

slim0801, 2020-11-22:
I also nees this.

slim0801, 2020-11-24:
Hi, i have made the integration, you can find the details here:
https://github.com/slim0801/esphome-MPL3115A2-sensor-integration

closed 2022-09-09 by kbx81