esphome/feature-requests #772: Support for Daikin BRC1H519W7 BLE controller

pacomm, 2020-06-13:
**Describe the problem you have/What new integration you would like**
Add support for Daikin BRC1H519W7 BLE controller
https://www.daikin.eu/en_us/products/BRC1H519W7.html

**Please describe your use case for this integration and alternatives you've tried:**
It would be use to control the air conditioner with a esp32 and integrate in Home Assistant

**Additional context**

pacomm, 2020-06-18:
When the the device is paired it's recognized as UE878 RF MODULE

pacomm, 2020-08-24:
A little light and many thanks to this man, but I don't have the necessary knowledge ...

https://github.com/blafois/Daikin-Madoka-BRC1H-BLE-Reverse

frigi83, 2021-08-14:
There is a python library to comunicate with the Madoka: https://github.com/mduran80/pymadoka 

The same guy as submitted a custom component to control the Madoka: https://github.com/mduran80/daikin_madoka the downside of this solution is that use the bluetooth of the host.

olbjan, 2021-11-10:
Amazing to see this, I would also appreciate if this was added to ESPHome as we have multiple Madoka Units spread over our office and it will be impossible to reach all of them from the central Home Assistant instance via Bluetooth.

mario-tux, 2023-03-29:
Something is moving... https://community.home-assistant.io/t/daikin-madoka-brc1h-bluetooth-custom-component/268594/80

Frank802, 2024-09-03:
UPDATE: there is a [pending PR](https://github.com/esphome/esphome/pull/5951)...