esphome/feature-requests #774: Allow configuration of base directory for builds

coogle, 2020-06-17:
**Describe the problem you have/What new integration you would like**

As it stands right now there doesn't appear to be a way to configure a base directory for the PlatformIO build files. I.e. if I run `esphome foo.yaml run` I end up with a `foo/` directory in PWD. This makes it hard to organize my device configurations cleanly and commit them to Git without committing PlatformIO stuff I don't want.

What would be nice is, at the very least, a command-line option against `run` and `upload` to specify the base directory for build output files:

`esphome --build-dir=/tmp/ foo.yaml run`

Which would use `/tmp/foo` for the build directory.

**Please describe your use case for this integration and alternatives you've tried:**

N/A.

**Additional context**

coogle, 2020-06-17:
Right after I opened this ticket I realized ESPHome ships with a `.gitignore` that actually does remove the build directories. But that said, it still seems like a good idea to provide a way to change the build-path so I'll leave this request open.

glmnet, 2020-06-17:
https://esphome.io/components/esphome.html

coogle, 2020-06-17:
Opps thank you. I missed that bit.

closed 2020-06-17 by coogle