esphome/feature-requests #776: External storage support

Hamper, 2020-06-21:
**Describe the problem you have/What new integration you would like**

Support for external storage like MB85RC256V instead of internal flash.

**Please describe your use case for this integration and alternatives you've tried:**

`esp8266_restore_from_flash` stores data to internal flash, but this flash has a limited number of write cycles. Why not to use external fram i2c modules like MB85RC256V with ~10^13 write cycles?

**Additional context**

ckuethe, 2021-01-10:
I was just about to make a feature request for this. I'd very much like to do this sort of thing using either a cheap/disposable i2c device to decouple the storage failure from the MCU, or using a virtually un-killable FRAM device.

I have killed the internal flash in a couple of 8266s with micropython and lots of state tracking, thus my switch to external storage.

nepozs, 2021-11-24:
I've found some resource about FRAM and EEPROM
https://easyswitch.pl/arduino-obsluga-eeprom-i2c/

tanishqmanuja, 2022-11-06:
any update or alternative to external eeprom?

ageurtse, 2022-12-15:
for me to, what is the status off this enhancement ?

nagyrobi, 2022-12-15:
No status, this was just an idea.

ageurtse, 2022-12-15:
> No status, this was just an idea.

to bad, trying to fidle out it myself :-) thanks for the fast reply

tanishqmanuja, 2022-12-16:
@ageurtse, Pls do post if you fidle something useful :)

ageurtse, 2022-12-25:
i'm slowly progressing in this, i'm making use of this nice [packages ](https://github.com/RobTillaart/FRAM_I2C/tree/master/examples)

nagyrobi, 2022-12-25:
Should be some kind of drop-in replacement for `esp8266_restore_from_flash` etc...

ageurtse, 2022-12-27:
i have a test version up and running, looks like it all worked fine. 
Maybee not the nices code, i'm a hobby coder but i mannaged it. 

https://github.com/ageurtse/ESP-Home-SO-counter-with-persistant-memory-module

tanishqmanuja, 2022-12-27:
Thanks @ageurtse, although the initial coding setup has much larger footprint than default restore_from_flash, because we have to make separate global var for all values we need. But definitely better than using onboard eeprom which has limited read/write cycles and cant be replaced.

ageurtse, 2022-12-27:
That was also my goal, a bilion writes without wearout. Even after reflashing or a power outage the memory is preserved even after several weeks.

nagyrobi, 2022-12-27:
What type of data does it supports? E.g. strings?

ageurtse, 2022-12-27:
not for this moment, but when it all is finished i will look into it. 
it is posible.

sharkydog, 2023-05-12:
And experimental component to write global preferences (such as persistent states) in FRAM.
https://github.com/sharkydog/esphome-fram/tree/test_pref
Currently it still has a lot debugging code, I am still testing this thing.

sharkydog, 2023-05-31:
Now in a PR esphome/esphome#4880, docs: esphome/esphome-docs#2950, [preview](https://deploy-preview-2950--esphome.netlify.app/components/fram.html)

CI checks made me completely transform the code, so there is barely anything in common left with the repo above.

tanishqmanuja, 2024-08-08:
Any update on how to use external i2c eeprom instead of fram ?

pilotak, 2024-08-08:
i recently made up EEPROM https://github.com/pilotak/esphome-eeprom

tanishqmanuja, 2024-08-08:
@pilotak great library, I had your repo already starred 😅. But i was looking for, what is the official way forward without this much boilerplate code ?

pilotak, 2024-08-08:
Then you need to make one, unfortunately. I needed such a feature, so for time being i came up with that

tanishqmanuja, 2024-08-08:
True 💯

pebblebed-tech, 2024-12-21:
I created and submitted a component last year to do this this see https://discord.com/channels/429907082951524364/1205893941367406592
and https://github.com/esphome/esphome/pull/6071
Unfortunately the esphome admins wanted a me to create a generic storage component and then derive a serial eeprom from it. I ran out of time as I only tinker during winter months