esphome/feature-requests #777: Support for MICS 6814

stskrp, 2020-06-22:
**Describe the problem you have/What new integration you would like**

Please add support for MICS-6814 sensor

**Please describe your use case for this integration and alternatives you've tried:**

Would be great to monitor CO VOC NH3 Nitrogen Oxygen Gas

**Additional context**

alexxed, 2021-02-21:
I'm testing this, it should work with the existing adc, but needs some fiddling to get ppm.
The values from the example are from https://forum.micropython.org/viewtopic.php?t=8349&p=47493, but it's weird that you would have to use different factors, so some docs reading is needed to get it right. And also figure out if it needs pull up resistors or not, like this article says: http://kstobbe.dk/2019/02/16/esp32-pms5003-bme280-mics6814-sensor-build/

Example:
```
sensors:
  - platform: adc
    pin: 32
    name: "CO"
    attenuation: 11db
    id: co
    update_interval: 60s
    filters:
      - lambda: return (x * (10000*(0.512/32767)));
  - platform: adc
    pin: 33
    id: nh3
    name: "NH3"
    attenuation: 11db
    update_interval: 60s
    filters:
      - lambda: return (x * (4000*(0.256/32767)));
  - platform: adc
    pin: 34
    id: no2
    name: "NO2"
    attenuation: 11db
    update_interval: 60s
    filters:
      - lambda: return (x * (80*(0.256/32767)));
```

andyjk15, 2021-08-16:
Did you get any further on getting the right values?
( Still not support for this sensor )

asucrews, 2021-12-24:
Just wondering if this start of support for this

https://github.com/MatejMecka/grove-multichannelgassensor-esphome/blob/main/mics-6814.cpp.h

johnboiles, 2022-02-06:
I looked into this a bit, specifically for the NO2 sensor. According to the datasheet, the OX sensor has a resistance from 0.8k-20k (my multimeter registered 12k so this checks out). It looks like this is resistance to ground. So it'd make sense that we'd need a pullup in order to register a voltage. Adding a 10k pullup to 3.3V would give a measurement range of 0.24V-2.2V which is a pretty good range to measure with the esp32 adcs (using `attenuation: 11db`).

But in practice I'm seeing this sensor spike up to 3.12V with the 10k pullup, which doesn't line up with max 20k resistance value for that sensor. The datasheet says "a single load resistor of 56 kOhms may be sufficient" so perhaps that's a more appropriate value (though I don't understand from the math why that'd be the case).

As for calibration, I'm not sure what to do. Maybe you can't do absolute calibration on these sensors. Perhaps the best I can do is establish a baseline measurement then look at changes. Does anyone have an intuition for how to think about establishing units when converting from voltage to PPM?

johnboiles, 2022-02-06:
These are times we cooked with our gas stove so it's definitely working. But not sure what the scale should be. But probably any detectable NO2 is not great.

[image]

closed 2022-07-01 by nagyrobi