esphome/feature-requests #779: Notification for Dallas "Requested conversion failed"

craigueh, 2020-06-23:
**Describe the problem you have/What new integration you would like**

I use ESPhome on a Sonoff 4ch R2 for Central Heating system with 6 ds18b20 temp sensors. 
Occasionally a sensor fails and takes down the lot, usually the error in the logs is "Requested conversion failed" but HA doesn't see any NaN and just gets stuck at the same temp indefinitely. 
This is a problem because the temp sensors are used to monitor boiler temp and hot water temp with automations to direct flow to radiators if hot water is over 60c.
What would be useful is an alert when ESPhome logs "Requested conversion failed" so I get notified on my phone and can take appropriate action.

Side note, I'm working on adding and external 5v supply to the Dallas sensors to hopefully fix this issue but on other ESP modules with a single temp sensor I've still seen the lock up once in a while.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

craigueh, 2020-06-24:
Update:

I managed to work around a notification with the following code to create a binary sensor of system alert state which can be used in Homeassistant to create a mobile app push alert. Works pretty well.

binary_sensor:
  - platform: template
    name: "Dallas Hub Alert"
    lambda: 'return ((App.get_app_state() & STATUS_LED_WARNING) != 0u);'

rradar, 2020-07-02:
I'm going to close this. If you are still in a need feel free open a new request

closed 2020-07-02 by rradar