esphome/feature-requests #780: Add support for the ac remote Fujitsu AR-JE5

danimart1991, 2020-06-23:
**Describe the problem you have/What new integration you would like**

The AC Remote Fujitsu AR-JE5 is not included in the list of compatible Fujitsu remotes of component climate fujitsu_general.

**Please describe your use case for this integration and alternatives you've tried:**

- I try to copy the codes using remote_receiver.
- I try the stable/next/dev versions to check if any contains the remote.
- I try the actual climate fujitsu_general component

**Additional context**

Maybe the SmartIR Broadlink integration could help to identify the codes: https://github.com/smartHomeHub/SmartIR/blob/master/codes/climate/1286.json

danimart1991, 2020-07-30:
News here. It seems that these remote models are the same:

- AR-DJ5
- AR-JE5
- AR-JE4
- AR-JE3
- AR-PV1
- AR-PV2
- AR-PV4
- AR-JE7
- AR-JE8
- AR-JE9
- AR-JE11

migabit, 2023-05-11:
Hello everyone. In the latest versions of ESPHome, I was able to read the codes for the AR-JE5 remote. It would be great if this somehow helps in the implementation of a full-fledged component for "IR Remote Climate"
[https://github.com/migabit/Fujitsu_IR_AR-JE5](url)

vicfergar, 2023-07-13:
I'm interested in it too. Did anyone here make any progress on that? I tried some codes contained in the project referenced by @migabit and works well (I tried the on/off  codes with a template switch)

vicfergar, 2023-07-14:
I made some progress modifying the existing `fujitusu_general` component to support this type of remotes. Changes can be found here: https://github.com/vicfergar/esphome/tree/feat/fujitsu-ar-je5

I'm using this board from [dudanov](https://github.com/dudanov/iot-uni-dongle) with the following config:
``` yaml
esphome:
  name: bedroom-climate
  friendly_name: bedroom-climate

esp8266:
  board: esp01_1m

external_components:
  - source: overriden_components
    components: [ fujitsu_general ]

# Enable logging
logger:
  baud_rate: 0
  #level: VERBOSE

# Enable Home Assistant API
api:
  encryption:
    key: "encryption_key"

ota:
  password: "ota_password"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Bedroom-Climate Fallback Hotspot"
    password: "ap_password"

captive_portal:

# IR remote transmitter (IR- pad connected to TSOP output of controlled device)
remote_transmitter:
  pin: GPIO13
  carrier_duty_percent: 100%

remote_receiver:
  id: rcvr
  pin:
    number: GPIO12
    inverted: true
  #dump: aeha

climate:
  - platform: fujitsu_general
    name: "Climatizador dormitorio"
    receiver_id: rcvr
```

Hope it helps!

mspiridakis, 2023-12-12:
Please add support for the Fujitsu AR-JE5. A lot of Fujitsu models use these codes.

mspiridakis, 2023-12-12:
> I made some progress modifying the existing `fujitusu_general` component to support this type of remotes. Changes can be found here: https://github.com/vicfergar/esphome/tree/feat/fujitsu-ar-je5 [...]

Newbie question: How do you properly reference the modified components in "external_components:" from https://github.com/vicfergar/esphome/tree/feat/fujitsu-ar-je5 ? 
Do I need to download the component folder locally ?

migabit, 2023-12-13:
@mspiridakis 

> Do I need to download the component folder locally ?

yes, look at an example of my config
`external_components:
  - source:
      type: local
      path: C:\1\esphome\components\
    components: [ fujitsu_general ]  `

@vicfergar 
Thank you for your work. I've been using the modification from your repository for several months and everything works great. I hope it gets into the main development branch.

mspiridakis, 2023-12-14:
Thank you for your time on this @migabit !

hldh214, 2024-02-07:
> I made some progress modifying the existing `fujitusu_general` component to support this type of remotes. Changes can be found here: https://github.com/vicfergar/esphome/tree/feat/fujitsu-ar-je5 [...]

Hello @vicfergar  , thank you for your work, it is greatly appreciated. I currently have a Fujitsu model designed for domestic use in Japan, with the remote control model "ar-rlb2j". I have tried the official "fujitsu_general" and "heatpumpir" libraries from esphome, as well as your modified version, but unfortunately, none of them can control my air conditioner. Therefore, I plan to follow @migabit 's approach and record all possible combinations of temperature and mode using a receiver. My question is, how can I convert these combinations into a cpp code similar to what you have written? Thank you.

alextessier6, 2024-06-05:
Thank you @migabit && @vicfergar !! Worked perfectly for a Fujitsu model ASU9CQ (remote model AR-JW28) 

Also, external components can be directly loaded from git which makes things a bit easier : 
```
external_components:
  - source: 
      type: git
      url: https://github.com/vicfergar/esphome.git
      ref: feat/fujitsu-ar-je5
    components: [ fujitsu_general ]
```

michaelcadilhac, 2025-02-11:
> Update: I found another repository that solved my problem: [crankyoldgit/IRremoteESP8266#2030](https://github.com/crankyoldgit/IRremoteESP8266/pull/2030) My yaml:
> 
> esphome:
>   includes:
>     - fujitsu_ir.h
>   libraries:
>     - https://github.com/shimmy-void/IRremoteESP8266.git#bce5c0b0eeda8431a95d95c9947e4008bb0b982f
> 
> # ...
> 
> climate:
>   - platform: custom
>     lambda: |-
>       auto fujitsu_ac = new FujitsuAC();
>       App.register_component(fujitsu_ac);
>       return {fujitsu_ac};
> 
>     climates:
>       - name: "Living Room Fujitsu AC"

Hey @hldh214 ; this certainly does not work out of the box like this anymore; how did you manage to get IRremoteESP8266 to be "compatible" with esphome?

hldh214, 2025-02-12:
> Hey [@hldh214](https://github.com/hldh214) ; this certainly does not work out of the box like this anymore; how did you manage to get IRremoteESP8266 to be "compatible" with esphome?

I don't understand your point, as I am still using this without any issues. Please provide more details about your situation.

michaelcadilhac, 2025-02-12:
> I don't understand your point, as I am still using this without any issues. Please provide more details about your situation.

Thanks for the prompt reply.  I'm simply wondering where you got `fujitsu_ir.h`; the only match on Google / GitHub is … this thread.  I adapted some similar files, but it's still buggy (not to mention that it seems [deprecated](https://esphome.io/guides/contributing#a-note-about-custom-components)). Cheers.

hldh214, 2025-02-13:
> Thanks for the prompt reply. I'm simply wondering where you got `fujitsu_ir.h`; the only match on Google / GitHub is … this thread. I adapted some similar files, but it's still buggy (not to mention [...]

Oh, it seems you're correct; it's being deprecated this year. I was following the [custom climate documentation](https://esphome.io/components/climate/custom.html) to build my `fujitsu_ir.h`, which simply calls functions from this pull request: https://github.com/crankyoldgit/IRremoteESP8266/pull/2030.

As the documentation states, after version "2025.1.0," my current configuration will indeed become unusable. Therefore, I think I should take the time to convert this into external components as suggested in the docs. I'll share my work once I finish it.

michaelcadilhac, 2025-02-13:
@hldh214 Great!  I took the time to go that route, as the custom climate doc seemed outdated in any case.  I can clean that up, but here's [the gist](https://gist.github.com/michaelcadilhac/06ac6ba2486e5a805c4d7167ed924d99).  Popped that into a `config/my_components/fujitsu_ir` folder, and used:

```yaml
external_components:
  - source: my_components

climate:
  - platform: fujitsu_ir
    name: "Fujitsu AC"
```

Quite a few things are hardcoded (model number, pin number, the fact that it's inverted, the capabilities of the unit, etc) and ideally they should be fetched from the yaml, but that's work for another day!