esphome/feature-requests #782: custom sensor platform has no "update_interval"

wjcarpenter, 2020-06-26:
**Describe the problem you have/What new integration you would like**

Many specifically supported sensors have a configuration item "update_interval". The `platform: custom` does not. Please add "update_interval" or the equivalent.
**Please describe your use case for this integration and alternatives you've tried:**

It seems that the default an only update interval for custom sensors is 15s. Sometimes that's OK, but sometimes it's not. The custom sensor example for the BMP085 can't have an update interval, but the later-implemented direct support for BMP085 does, like most of the other sensors.
**Additional context**

Perhaps there is some lambda way to set the update interval. If so, I'd be glad to be pointed to it.

glmnet, 2020-07-02:
try this:

````
class MyCustomSensor : public PollingComponent, public Sensor {
 public:
  // constructor
  MyCustomSensor(uint32_t update_interval) : PollingComponent(update_interval) {}

  void setup() override {
    // This will be called by App.setup()
  }
  void update() override {
    // This will be called every "update_interval" milliseconds.
  }
};
````

...

```` 
sensor:
- platform: custom
  lambda: |-
    auto my_sensor = new MyCustomSensor(5000);
````

wjcarpenter, 2020-07-02:
Nice. Thanks, @glmnet

closed 2020-07-04 by glmnet