esphome/feature-requests #783: Input_number entity definition

vicfergar, 2020-06-26:
**Describe the problem you have/What new integration you would like**

I would like expose an input_number entity using ESPHome. It can be done using a template to indicate what will happen "on_value"

**Please describe your use case for this integration and alternatives you've tried:**

I would like to implemente a volume and bass level control of a speaker set using ESPHome. Currently I'm using monochromatic light to expose volume and bass level as dimmable lights. I would like to expose two input_number entities instead.

**Additional context**

CornHead764, 2020-12-11:
I too would like to see this option available. I'm wiring an esp32 into a dumb video switcher, allowing for switching between it's 4 inputs through esphome/home assistant. I'll be exposing them as momentary switches to HA, but it would be nice if it could be exposed as a media_player with the inputs as a source_list.

kquinsland, 2021-02-17:
I have two use cases for this:

- I have attached a servo to the volume knob and a relay to the power switch for a set of speakers. Currently i have ESPHome subscribe to  a topic and parse out integers to drive the servo and a basic GPIO switch for the power button. I am considering wrapping them in something like this: https://github.com/Sennevds/media_player.template

- I would _really_ like to hook a basic speaker up to the i2s interface (see: https://github.com/esphome/feature-requests/issues/599) so i can have Home Assistant send audio streams to the speakers. This would work for media, but my intended use case would be for simple TTS streams like "5 min remaining" or "it's past 1 PM and you still haven't loaded the dishwasher. Do that now" ... for example.

Basically, i'd like to use the ESP32 / i2s interface as a budget / less power hungry version of a cheap/low-powered ARM based device for running [snapcast](https://github.com/badaix/snapcast)

DieKatzchen, 2021-02-21:
I'd like to second this issue. My use case is that I have 3 displays and a number of inputs running through a series of matrix switches for which I was going to have to expose a whole bunch of momentary switches and then write a whole bunch of glue logic on the HA side. With this feature I can simply select the input I want and the esphome can send all the appropriate remote codes.

nagyrobi, 2022-07-01:
https://esphome.io/components/media_player/i2s_audio.html

glyndon, 2022-07-30:
This is a great suggestion.
This component doesn't have to *be* a media player, just look like one.
Then those of us with diverse media control needs could adapt the template to control whatever [media-playing] devices we attach our automata to.

landonr, 2022-09-21:
i manage a bunch of media players in my project here. here are my key features
- control play, pause, next, previous, stop, shuffle, repeat, mute
- player state: playing, paused, idle, stopped
- load source list from home assistant. sonos is listed as a separate entity i think
- set source from list
- player type: tv, speaker
- grouping media players
- control treble, bass

nice to haves
- playlist name
- queue position and count
- control multiple media players
- proportional volume control on grouped speakers
- realtime media position. it seems like the position in home assistant is delayed and uses `media_position_updated_at` for reference
- set soundbar media player as tv soundbar (to get volume %)
- custom switches eg sonos night sound, speech boost, loudness