esphome/feature-requests #784: ESPHome support for SONOFF iFan03 (Wi-Fi Ceiling Fan And Light Controller) by ITead

Hedda, 2020-06-26:
**Describe the problem you have/What new integration you would like**

Please consider adding support for SONOFF iFan03 Wi-Fi Ceiling Fan And Light Controller from ITead in ESPHome.

https://www.itead.cc/sonoff-ifan03-wifi-ceiling-fan-light-controller.html

**Please describe your use case for this integration and alternatives you've tried:**

SONOFF iFan03is supported in Tasmota

https://tasmota.github.io/docs/devices/Sonoff-iFan03/

**Additional context**

Summer is here! ;)

exeljb, 2020-07-09:
I too would like to see this added to esphome

elfhack, 2020-08-02:
Same here. It's not that adapting ifan02 would be a problem - the pins are already known as this device works nicely with tasmota. The biggest hurdle is the RF433 remote interface, which now speaks serial, instead of relaying the pressed buttons through gpio

fmsalinas, 2020-08-18:
me too. is the only device remaining to migrate to ESPHome!

sliwma, 2020-10-25:
I have a working config, gonna share it tomorrow. Took me about 20 minutes to make it - based on iFan02 config.

elfhack, 2020-10-25:
@sliwma And the rf interface?

sliwma, 2020-10-25:
Yup. I need to readd light switch, as I don't use it.

Z wyrazami szacunku
Maciej Śliwiński

sliwma, 2020-10-26:
I'm gonna add it to the cookbook. But for now...
https://github.com/sliwma/mysmarthome/blob/main/iFAN03.yaml

sliwma, 2020-10-26:
And buzzer support:
https://github.com/sliwma/mysmarthome/blob/main/ESPHome%20configs/Sonoff-iFAN03/Sonoff_iFAN03_buzzer.yaml
Works as it was out-of-box I guess - can be toggled with upper right button and with HA switch, beeps once/twice/thrice, depending on speed.

CPT-Neptune01, 2021-01-02:
Thanks @sliwma for the ESPHome code, I have used a modified version to implement my own custom iFan03 setup using GPIO5 as a switch input to control the light from a wall switch.
Can the bottom 2 buttons on the remote be used for ESPHome functions? If so what are the RF codes for these buttons?

sliwma, 2021-01-02:
Hey, I tried to dump raw values but I didn't receive any code while hitting these buttons.

CPT-Neptune01, 2021-01-02:
Thanks for the response. I will give it a try, holding the button is not an issue for the function I want as it is something that would be better to not be easily bumped on or off.
> According to manual, left one is used to pair remotes.
>
That makes sense that there would be no code sent to the ESP chip then.

ishioni, 2021-04-07:
@sliwma Hi. I'm about to migrate my tamota ifan03 to your config, but I have 2 questions
- Is there any specific reason you're using a template switch instead of a generic input switch for the buzzer?
- I'm using the light switch to instead drive an oscillation motor - https://gist.github.com/ishioni/1c45ab1faf80d0deef4de049cf437dcf Does this look correct? Any idea how to check if the fan is running before enabling oscillation - there is no fan.is_on function in esphome :/

Ethan-G, 2021-07-03:
@sliwma I just got an iFan03, was looking around for ESPHome code and found this. Works brilliantly, thanks very much for your efforts! I assumed I was going to have to write that myself from the Tasmota mappings I found.

evanheckert, 2021-09-29:
Used ESP Flasher to install the bin created by ESPHome, and it flashed successfully. Would you expect that after a power cycle it should successfully connect to your network while still plugged into via USB? It's not showing up on the network.

Ethan-G, 2021-09-29:
@evanheckert Is your USB connection causing it to go into flash mode? If unsure, try just plugging it in to 230v like it would be when installed.

evanheckert, 2021-09-30:
Good question @Ethan-G I'm not sure. It's an iFan04, and going in the nursery so I need to minimize amount of time with the switch socket open, which I believe is the only way to power it if USB won't work. I did try powering it with only the ground and 3.3V pins, but it doesn't seem to be connecting or broadcasting the backup SSID that's in the config. I reflashed it with ESPHome and it says it succeeded, but I'm not seeing a connection after that.

Ethan-G, 2021-09-30:
@evanheckert  Well to check if anything's working, if it's not connected to your WiFi then it should be creating its own fallback network (if it's configured to do so; if not, it's a good idea).

You can power it on your desk by just getting an old power cable, cutting the end off and wiring it up using that. If you're not comfortable doing that then.. well what are you playing with 230v devices for? ;)

Hang on, the iFan is meant to be a replacement for an in-fan controller. If you're able to control it from the wall, then a smart wall fan controller is probably a better idea surely. If you're in Australia or New Zealand I recommend [this](https://www.bunnings.com.au/deta-grid-connect-smart-fan-speed-controller-with-touch-light-switch_p0098815).

Eric-boop, 2021-12-25:
@sliwma do you think your code will work for the iFan04-L US version? The 04 is supposed to be the same as the 03, with us different capacitors of 120v.

JonLaliberte, 2022-06-06:
Looks like iFan04-L compatible code can be found here: https://community.home-assistant.io/t/sonoff-ifan04-esphome-working-code/323402/21

Would be nice to get it merged in to ESPHome though.

closed 2022-07-01 by nagyrobi