esphome/feature-requests #785: ble tracker works with smartphones

xcray, 2020-06-27:
esp32 ble tracker not compatable with smartphones
**Describe the problem you have/What new integration you would like**

current esp32 ble tracker not works with smartphones, because the modern phones always adv random addresses.
**Please describe your use case for this integration and alternatives you've tried:**

with compatibility with smartphones, the worth of esp32 ble tracker could be higher, for sure. 
**Additional context**

andrewjfreyer has a wonderful shell script for ble presence detecting, using name querry, pls check on:
https://github.com/andrewjfreyer/monitor
so, how about porting to esp32?

krambriw, 2020-07-02:
I have been running Andrew's shell script some years now in a RPi3 with external BT Dongle, it has been working excellent so yes, would be nice to have it in an esp32

markusressel, 2020-07-06:
While I very much support this request, I think this might be a duplicate of #481

xcray, 2020-07-07:
> While I very much support this request, I think this might be a duplicate of #481

Sure! the same issue. thanks for your comment.

closed 2020-07-07 by xcray