esphome/feature-requests #787: Add "invert" option to gate_pin on ac_dimmer component

tracestep, 2020-06-28:
I am testing the `ac_dimmer` component on the dev branch from [https://github.com/esphome/esphome/pull/880](url) (pull request is closed for comments) and it works well, but it would be useful to add some kind of "invert" option for the triac `gate_pin`, so that it is activated on logical low instead of logical high. Maybe that could be implemented by `method: leading_pulse_inverted` or `inverted: yes` option for the output / gate pin.

This is needed to use GPIO0 or GPIO2 to control triacs with ESP-01 boards, because with an active high logic the optocouplers would pull those IO lines to GND, causing boot to fail. GPIO3 (TX) on those boards also would activate the triac on boot with the default active high logic.

I am sure this would be trivial to add, but I am new to ESPhome and not yet familiar with its C++ mechanics (or Git for that matter) to try it myselff.

glmnet, 2020-07-01:
Gate pin uses pin schema, that is, configuring it’s output as inverted should work that way. 

If it does not work please fill a bug report. 

I’m closing this but keep asking if you need more info.

closed 2020-07-01 by glmnet

tracestep, 2020-07-01:
I was not sure if it was supposed to... Thanks for the answer, bug reported submitted: [https://github.com/esphome/issues/issues/1318](url)