esphome/feature-requests #788: DFPlayer Mini additional features

jscarlett5, 2020-06-29:
**Describe the problem you have/What new integration you would like**

The DFPlayer has many unused features. There are a number of query commands that could provide useful feedback of the player's status. Right now, I send commands (which is amazing! thank you!), but I would love for the unit to report volume level, playing status and if possible track name to home assistant.

**Please describe your use case for this integration and alternatives you've tried:**

I would like to display status details of the device on my dashboard. 
Based on [https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299](url)
-volume level (Serial 0x43)
-current status (Serial 0x42)
-playback mode (Serial 0x45)
-track name (if possible. Serial 0x4B?)

**Additional context**

glmnet, 2020-07-01:
Track name would be interesting if provided. I couldn't find any docs pointing to that. Where did you find info on the 0x4B command?

jscarlett5, 2020-07-01:
> Track name would be interesting if provided. I couldn't find any docs pointing to that. Where did you find info on the 0x4B command?

jscarlett5, 2020-07-01:
> Track name would be interesting if provided. I couldn't find any docs pointing to that. Where did you find info on the 0x4B command?

It's in https://wiki.dfrobot.com/DFPlayer_Mini_SKU_DFR0299 under Serial Query Cmd. It says "Queries the current track of TF card" But maybe that means track number or something. They didn't give an example of it.

closed 2020-07-01 by jscarlett5