esphome/feature-requests #791: Add CJMCU-8128 support

riochicken, 2020-07-05:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

idavydov, 2020-11-08:
CJMCU-8128 is three different sensors on the same board:

- CCS881 (eCO2 & TVOc)
- BMP280 (temperature & pressure)
- Si7021 (temperature & humidity)

All of them are already supported by ESPHOME (amazing, I'm really impressed).

Here is the config which works for me:
```
sensor:
  - platform: bmp280
    temperature:
      name: "Temperature (BMP280)"
      id: bmp280_temperature
      oversampling: 16x
    pressure:
      name: "Pressure"
    address: 0x76
    update_interval: 60s
  - platform: htu21d
    temperature:
      name: "Temperature (Si7021)"
    humidity:
      name: "Humidity"
      id: si7021_humidity
    address: 0x40
    update_interval: 60s
  - platform: ccs811
    eco2:
      name: "CCS811 eCO2 Value"
    tvoc:
      name: "CCS811 Total Volatile Organic Compound"
    temperature: bmp280_temperature
    humidity: si7021_humidity
    address: 0x5A
    update_interval: 60s
```

Please note that:
- Temperature and humidity parameters for ccs811 are not necessary but are supposed to correct for changes in humidity and temperature.
- It [seems](https://mindeon.com/posts/coding-journal/cjmcu-8128-code-example/) that some versions of CJMCU-8128 have HDC1080, not Si7021. You need to change your settings accordingly. There might be also other differences, e.g., different addresses.
- You need to take special care calibrating the sensor. First, run it for 48 hours. Then you need to set the baseline (see [here](https://esphome.io/components/sensor/ccs811.html), ideally somewhere in the fresh air). Finally, they recommend a periodic baseline correction, which I am not sure is implemented in the esphome module for CCS881.
- I have an impression that CCS881 is not a very precise sensor; good CO2 sensors are expensive.

christiaangombert, 2021-04-15:
To build up on the work of @idavydov as posted above, I made some minor adjustments to the configuration:

`i2c:
  sda: 21
  scl: 22

sensor:
  - platform: bmp280
    temperature:
      name: "Temperature (BMP280)"
      id: bmp280_temperature
      oversampling: 16x
    pressure:
      name: "Pressure (BMP280)"
      id: bmp280_pressure
    address: 0x76
    update_interval: 60s
  - platform: hdc1080
    temperature:
      name: "Temperature (HDC1080)"
      id: hdc1080_temperature
    humidity:
      name: "Humidity (HDC1080)"
      id: hdc1080_humidity
    address: 0x40
    update_interval: 60s
  - platform: ccs811
    eco2:
      name: "eCO2 (CCS811)"
      id: ccs811_eco2
    tvoc:
      name: "Total Volatile Organic Compound (CCS811)"
      id: ccs811_tvoc
    temperature: bmp280_temperature
    humidity: hdc1080_humidity
    address: 0x5A
    update_interval: 60s
`

closed 2022-07-06 by nagyrobi