esphome/feature-requests #792: BLE Bridge

anthonyangel, 2020-07-05:
**Describe the problem you have/What new integration you would like**
I'd like to be able to use devices controlled by esphome as BLE bridges (to connect BLE devices to a local network)

**Please describe your use case for this integration and alternatives you've tried:**
I currently use https://github.com/bendikwa/igrill to monitor a Weber iGrill (on a raspberry pi, I'd like to be able to use an esp instead to read from the sensors. This device would be read-only, but there are some which would benefit from read/write functionality (e.g. Anova Sous Vide machine)

**Additional context**

anthonyangel, 2021-06-18:
With the new(ish) BLE Client Sensor component I've started poking this.

So far I can get the battery status with this config:
```
sensor:
  - platform: ble_client
    ble_client_id: igrill
    name: "iGrill battery level"
    service_uuid: '180f'
    characteristic_uuid: '2a19'
    icon: 'mdi:battery'
    unit_of_measurement: '%'
 ```
 
The debug logs show me that the device has 2 other services, which have 7 and 9 characteristics. I've tried taking the UUIDs from the logs (I don't know why but the logs showed malformed UUIDs and I had to move the dashes around), but this didn't give me any valid data.

From reading various python repos (like the one linked in the top post) it looks like the values are encrypted with a static key, so I'm going to try and experiment with a lambda to decrypt the values. So far I've tried the code for [raw data parsing](https://esphome.io/components/sensor/ble_client.html#raw-data-parsing-lambda), just to get the values, but it still errored.

anthonyangel, 2021-06-18:
For reference, here's the rest of the config (which currently is useless, but might help someone):
```
  - platform: ble_client
    ble_client_id: igrill
    name: "1"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0001-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "2"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0002-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "3"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0003-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "4"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0004-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "5"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0005-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "6"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0006-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "7"
    service_uuid: '64AC0000-4A4B-4B58-9F37-94D3C52FFDF7'
    characteristic_uuid: '64AC0007-4A4B-4B58-9F37-94D3C52FFDF7'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "8"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0001-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "9"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0002-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "10"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0003-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "11"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0004-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "12"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0005-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "13"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0006-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "14"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0007-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "15"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0008-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

  - platform: ble_client
    ble_client_id: igrill
    name: "16"
    service_uuid: 'A5C50000-F186-4BD6-97F2-7EBACBA0D708'
    characteristic_uuid: '06EF0009-2E06-4B79-9E33-FCE2C42805EC'
    lambda: |-
      return *((float*)(&x[0]));

frank-ydf, 2021-08-17:
I'm very interested too.

closed 2022-07-06 by nagyrobi

ronaldt80, 2022-07-13:
Hi @nagyrobi , is there any guidance/experience on implementing this for the igrill/Weber sensor probes?