esphome/feature-requests #793: Version history

malogic, 2020-07-06:
**Describe the problem you have/What new integration you would like**

when i create a new device i write some code and gett the basic functionality working.
i then add features to that code.
when i'm done with a implementation of a feature and have tested that all is working i like to save of the code into a version so that i always can go back to the old working code when i have added a new feature that is not working.

now i have a separate notepad++ dockument to save the older versions.

**Please describe your use case for this integration and alternatives you've tried:**

my idea is to add a button next to the save button "Save as New version"
and a dropdown in the editor view where you can load old versions.
a delete button to be able to remove older versions that is not needed any more.  

**Additional context**

glmnet, 2020-07-06:
Integrating a SCM (Source code management) tool on ESPHome dashboard will be a bit overkill.

You can try using git for that, if you are using the hass supervisor, then take a look at the File Editor add on https://github.com/home-assistant/hassio-addons/tree/master/configurator

You can use that right now to push your configs to a github account (you can start with a private repository and is up to you to share your configs later)

OttoWinter, 2020-07-15:
Yeah I agree, you should use some version management system for that, for example git. For example you can push your configs to some git repo and pull it on the Hassio machine, or you can mount the config folder via SMB and edit the files with VSCode.

closed 2020-07-15 by OttoWinter