esphome/feature-requests #794: IOS app (Hamburger menu) back feature request

passie, 2020-07-06:
**Describe the problem you have/What new integration you would like**

Using ESPHome in combination with the Home-assistant add-on and the IOS app. I would like to use the hamburger menu to go back or select a different menu item. 

[image]

for example HACS (a popular add-on) does have this menu icon and i'm able to go to a different menu item. 
[image]

ESPHome is missing this menu icon.
[image]

**Please describe your use case for this integration and alternatives you've tried:**

Currently i'm unable to use ESPHome in the IOS app and go back to different menu item's if needed.

**Additional context**

jonathanadams, 2020-07-16:
Does swiping from the left not bring up the menu in the HA app?

passie, 2020-07-16:
Omg, you are right. Sorry

passie, 2020-07-17:
Not a issue, since it's possible to swipe from the left.

closed 2020-07-17 by passie