esphome/feature-requests #797: TOF10120 Time Of Flight Distance Sensor up to 180cm

Frederik-V, 2020-07-08:
Dear.

It would be great, next to the VL53L0X Time Of Flight Distance Sensor, to have the TOF10120 as well.

Datasheet:
[https://github.com/simpleiot/reference/blob/master/sensors/TOF10120_english.pdf]

This sensor is also based on the time of flight principle and can communicate either with RS232 (UART) and I2C.
It can range up to 180cm and has a sealed sensor body with screws. Very useful in moistured areas (rainwell).

Thank you in advance for the efforts to implement/add this sensor.

Kind regards,
Frederik

closed 2021-04-28 by jesserockz

Frederik-V, 2021-04-29:
Dear.

I saw that this item has been closed.

Where can I find the (final) solution on esphome.io ?
I don't find any reference to the TOF10120.

Next to the I2C solution, is the RS232 also be incorporated?

Thank you in advance for your answer.

Kindest regards,
Frederik

jesserockz, 2021-04-29:
Hi @Frederik-V 

Issues and feature requests are closed when PRs are merged to dev. 

If you install the dev version of ESPHome then you can use this now. Docs are at https://next.esphome.io

Frederik-V, 2021-04-30:
Dear @jesserockz.

Thank you for your quick response on my comment and all the efforts so far on this item.
Meanwhile I have explored the DEV version and documentation for the respective sensor.

I was wondering if the RS232 (UART) communication can be integrated as well.
Unless I'm mistaking, but for so far only the I2C communication is available.

The advantage of the RS232 communication is that the distance from the sensor to the controller can be much larger. 
That means that the controller can be in house, while the sensor module (TOF) can be outside.
If you want to use the sensor (sealed) in a humid environment, the controller can be placed away for that humid environment.

Thank you in advance to take my question/request into consideration.

Kindest regards,
Frederik

AGordiGuerrero, 2021-05-27:
Hi everyone.

I have been working these days with the TOF10120 sensor and I managed a lot of things.

You can take a look at: https://github.com/AGordiGuerrero/TOF10120_I2C

I hope it helps. It is only the beginning... I have then working in groups without problems using I2C. I will upload more codes soon.

Best wishes.
A.Gordillo