esphome/feature-requests #799: Non-blocking http requests

marianomd, 2020-07-09:
**Describe the problem you have/What new integration you would like**

It would be nice that http requests wouldn't block. If you have a Polling component, for example for fading a led, when a http_request is waiting for response, the PollingComponent is not called until the request finishes or times out.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

miceno, 2025-01-28:
I am planning to start working on this, by integrating [AsyncHTTPRequest_Generic](https://github.com/khoih-prog/AsyncHTTPRequest_Generic) library. This library only supports HTTP, but not HTTPS. There is another version for HTTPS (see [here](https://github.com/khoih-prog/AsyncHTTPSRequest_Generic)).

 I had a look at other libraries, but some are just prototypes or lack support ([ESP8266AsyncHttpClient](https://github.com/jstrausd/ESP8266AsyncHttpClient), [ESPAsyncHTTPClient](https://github.com/hatlabs/ESPAsyncHTTPClient)), or only support ESP32 (). I pretend to create one for ESP8266. Should you know about any other library, please, let me know.

Concerning the implementation, I am considering splitting it into two parts: one for the sending the request and another for receiving the response. There is a challenge on how to implement it as a component, and it is related to the way the actions are created, I am not an expert on espHome internals, but I think it is not prepared for actions that start and keep running on the loop. I will appreciate any insight on how to do it properly or pointing me to an sample implementation.

Anyway, is there anybody else interested?

phillipthelen, 2025-03-18:
I was surprised when I started using the http_request module to find that it was in fact blocking. Having a non-blocking way for making http(s) requests would be very nice!