esphome/feature-requests #803: Markdown

meijerwynand, 2020-07-10:
**Describe the problem you have/What new integration you would like**

_"In my opinion, Markdown would have been the much better choice in hindsight.."_
https://esphome.io/guides/contributing.html

I have contributed to documentation in the past, however, RST is just not something I use or seem to get use to. 

I am of the view that changing the documentation to a more popular / frequently used format, it can potentially assist in additional contributions by either means of edits/modifications on existing pages and/or new contributions such as cookbooks or configs.

**Please describe your use case for this integration and alternatives you've tried:**

Looking further into the matter, I attempted a mirror on the site via ```wget``` and bulk html to md conversion using ```pandoc```. For rendering I looked at [picocms](http://picocms.org/). It is not a clear-cut copy/paste and done task. That also excludes the styling (to be more inline with the current look and feel) and the search functionality. There are a number of factors and hours involved to "convert to markdown" for the documentation. 

**Additional context**

What would you consider as a good or acceptable proof of concept for and markdown conversion (if any at all)? 

Looking forward to your thoughts on the matter.

Thanks

OttoWinter, 2020-07-15:
I don't think anymore that RST was really that bad of a choice.

Yes it's not friendly for beginners, but it allows us to add lots of primitives like `:apiref:` (automatic link to documentation) for example.

But the best feature by far is cross-document linking. In markdown, everytime you want to link to another document or section (for example to document the type of config options) you'd have to paste a full link in markdown.

With RST and sphinx, we can easily create cross-document links that also ensure there are no dead links etc.

I think a lot could already be done by improving the documentation on the contributing page. Realistically, most people only use a tiny subset of RST when writing docs anyways and you can just copy that from a similar file that already exists.

meijerwynand, 2020-07-28:
Thanks for the feedback and for sharing your views. 

I infer from your response you are happy to keep the documentation as is using RST as explained above. 

Regards

closed 2020-07-28 by meijerwynand