esphome/feature-requests #804: Show if current YAML is running on ESP

cromulus, 2020-07-10:
**Describe the problem you have/What new integration you would like**
Right now, I don't know what version of my configuration YAML is running on the ESP device. It would be helpful to have a log of deploys and/or a display of the currently running YAML's SHA1, or something.

**Please describe your use case for this integration and alternatives you've tried:**

A bit of version control would be nice, as well as visibility into the currently running YAML.

**Additional context**

aartokke, 2020-07-10:
for that reason i'm using a template text sensor with a version number and upload date to see if i have already uploaded a changed yaml. Because I kept forgetting after a few days... "did i already uploaded it??"

allan-reynolds, 2020-09-23:
Its not ideal, but I use the version text_sensor for this.   I can compare the running build time with yaml file modification timestamp quite easily. 

https://esphome.io/components/text_sensor/version.html