esphome/feature-requests #807: Support for i2c multiplexer tca9548a

guestisp, 2020-07-11:
tca9548a is an i2c multiplexer, allowing to create 8 i2c busses from a single bus

Really useful when you can't change the remote i2c address and you have to connect multiple devices with the same address

Implementation seems pretty simple, to talk to one bus,you just have to send a byte with the bus number and following communicatons are routed automatically

mkuziel, 2020-10-22:
What are the limitaitons? Basically it looks as simple issue, any experience person can comment on it?

Primo83, 2020-10-23:
It would be fine to connect for example a lot of bme280 to one central esp32 (not only 4)

andreashergert1984, 2020-12-06:
We need somebody for QA...

Casino, 2021-01-24:
Any news for that feature?

andreashergert1984, 2021-01-25:
Actually i am waiting for the merge. as there is imho no open point left.
correct if there is something to be done.

andreashergert1984, 2021-01-27:
FR is tagged for merge in next Release: https://github.com/esphome/esphome/milestone/38

closed 2021-03-29 by glmnet