esphome/feature-requests #808: Support for MHO-C401 Smart BT Thermometer Hygrometer (E-Ink)

XantXant, 2020-07-14:
**Describe the problem you have/What new integration you would like**

Support for MHO-C401 Smart BT Thermometer Hygrometer (E-Ink).

**Please describe your use case for this integration and alternatives you've tried:**

My use case is to integrate the MH-C401 over esphome into HomeAssistant.

**Additional context**

Link to the sensor:
[https://www.banggood.com/Upgrade-Version-MMC-E-Ink-Screen-BT2_0-Smart-Bluetooth-Thermometer-Hygrometer-Works-with-MIJIA-App-Home-Gadget-Tools-from-Xiaomi-Youpin-p-1696154.html?rmmds=myorder&cur_warehouse=CN](url)

XantXant, 2020-07-14:
It looks like it is the same encryption like in the LYWSD03MMC sensor.
Following issue: #552

nikito7, 2020-11-12:
If someone needs. Vevs fork merged with current dev.

https://github.com/nikito7/esphome_vevs_merge

- atc_mithermometer
- lywsd02 battery
- mho-c401

pizipi, 2021-01-19:
> If someone needs. Vevs fork merged with current dev.
> 
> https://github.com/nikito7/esphome_vevs_merge
> 
>     * atc_mithermometer
> 
>     * lywsd02 battery
> 
>     * mho-c401

Hi,
I am interested to use this for my mho-c401, but sorry I am a noob, I already have esphome fhashed to my esp32 from docker intsallation, which looks differents from this one called vevs. Would you tell me how to use it?

nikito7, 2021-01-19:
Just use last "esphome dev"

pizipi, 2021-01-19:
> Just use last "esphome dev"

This is what I already tryed from docker-hub, I choose esphome with dev tag instead of latest, so I got the esphome dev I guess. But when I configure a "platform: xiaomi_mho-c401" it says "platform not found"

nikito7, 2021-01-19:
```
  - platform: ble_rssi
    mac_address: "A4:C1:38:4Dxxxxx"
    name: "T2 MHO RSSI"

  - platform: xiaomi_mhoc401
    mac_address: "A4:C1:38:4D:xxxxx"
    bindkey: "9915a51dbedcb804a685a7d0e85fxxxx"
    temperature:
      name: "T2 MHO Temp"
    humidity:
      name: "T2 MHO Hum"
      accuracy_decimals: 0
    battery_level:
      name: "T2 MHO Batt"
```

pizipi, 2021-01-19:
This is what i put in my yaml config file:
 - platform: xiaomi_mhoc401
    mac_address: "A4:C1:38:10:xx:xx"
    bindkey: "c935915941217da560d0071a0ea97da2"
    temperature:
      name: "Sejour_mho Temp"
but still have the same error, "platform not found" for sensor.xiaomi_mhoc401
thank you for your help

my esphome is v1.17.0-dev  from: docker pull esphome/esphome:dev

nikito7, 2021-01-19:
My mistake

I dont know how docker works

Pip:
```
pip remove esphome
pip install https://github.com/nikito7/esphome_vevs_merge_2/archive/dev.zip
```

pizipi, 2021-01-19:
> My mistake
> 
> I dont know how docker works
> 
> Pip:
> 
> ```
> pip remove esphome
> pip install https://github.com/nikito7/esphome_vevs_merge_2/archive/dev.zip
> ```

So thank you very much for your help, it work now! :)
So your git is different from the one on the docker hub

closed 2021-01-27 by jesserockz

nikito7, 2021-01-28:
Use last esphome dev now