esphome/feature-requests #812: Add native platform target

OttoWinter, 2020-07-16:
**Describe the problem you have/What new integration you would like**

A new [`native` / `host`](https://docs.platformio.org/en/latest/platforms/native.html) platform target to compile for.

**Please describe your use case for this integration and alternatives you've tried:**

Currently, the only way to test/debug ESPHome is to compile and upload it to an ESP. This incurs long wait times, makes debugging much harder etc.

If we could compile ESPHome to a *native* platform target (so directly for the host OS), then testing and debugging could become *much* easier. Have a bug? Just launch `gdb` or even visual debuggers and step through the program without buying/wiring up hw debugger.

Obviously this only helps for logic issues, and not timing/hardware related issues.

**Additional context**

PlatformIO [already has support for it](https://docs.platformio.org/en/latest/platforms/native.html), so we even wouldn't need to write our own tool.

Of course this would be a significant effort because we need to mock all Arduino-framework functions, but that could be quite beneficial in the long term because it forces us to write better hardware abstraction layers.

(I have no intention of implementing this in the near future, just an idea I had)

OttoWinter, 2020-07-16:
Thinking further: This could easily extended to work with raspberry Pis.

I'm not 100% sure what the use case for that would be though except maybe to get ESPHome's YAML syntax to work.

glmnet, 2020-07-16:
This lines up for the people who is trying to add other micro controllers like the STM32 etc. https://github.com/esphome/feature-requests/issues/687

bwims, 2023-01-28:
Now that there is support for the RP2040, is there a better reason to reject support for the BL602 than "It's not ESP"?
https://github.com/esphome/feature-requests/issues/1049

The MagicHome Led Controller was incredibly useful until they switched from ESP to BL602.  I'm sure a lot of people would be delighted if you were able to support it.  Similarly the Realtek processors used by light bulbs now.