esphome/feature-requests #813: Tutorial for creating a native component from a custom component

ei-ke, 2020-07-16:
**Describe the problem you have/What new integration you would like**

As I have very basic programming knowledge I'd love to see a tutorial how to create a "native component" from a working custom component that I e. g. created using existing Arduinio libraries so that it could be used by other who have NO programming knowledge.

I have no idea if this is feasible or if it would just be too much for a tutorial.

In my case I created custom components for VEML6075 and MAX44009 and would like to add them to the ESPHome repertoire.

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

glmnet, 2020-07-18:
There is no one but it's no so hard

Components are usually a python (__init__.py) and a c header file and a cpp source file. 
The name of the folder gives the name to the component, so it's important, knowing this you can already find the component most similar to the one you want to implement, copy it in a new folder and start replacing where it makes sense.

glmnet, 2020-07-18:
However for components we highly discourage the use of third party libraries, this is so because it adds dependencies and makes overall maintain harder.

ei-ke, 2020-07-18:
Modifying an existing component as a new one would also have been my approach.

I understand the issue with libraries. Are you commonly doing a complete rewrite based on the data you have for a component? I personally would ask the original author for permission/check if his license is compatible and just leave a hint in the component that it's based on X. Though I prefer asking instead of just doing :)