esphome/feature-requests #815: Binary sensor: on_long_press automation

TheLastGimbus, 2020-07-21:
**Describe the problem you have/What new integration you would like**

One automation that I miss in binary sensor is `on_long_press`

**Please describe your use case for this integration and alternatives you've tried:**

You can use `on_multi_click`, and configure it to your needs...

**Additional context**

...but it would be nice to have it already in place.
If someone told me where is this in code and how should I do it, I could totally make a PR for that :)

glmnet, 2020-07-21:
it's quite easy with `on_click`

https://esphome.io/components/binary_sensor/index.html#binary-sensor-on-click

glmnet, 2020-07-21:
actually you're right, `on_multi_click` is even better because you don't have to release it....

```
on_multi_click:
- timing:
    - ON for at least 5s
  then:
    - logger.log: "Pushed for 5 seconds"

closed 2022-07-06 by nagyrobi