esphome/feature-requests #816: Adding time-based options to cover template

erasmuswill, 2020-07-21:
**Describe the problem you have/What new integration you would like**

I'd like to have a time-based cover that gets its state verified at each end.
**Please describe your use case for this integration and alternatives you've tried:**

I would like for my template cover to provide additional close percentage information to HASS.
**Additional context**

I have no problem with implementing this but I will need a bit of guidance on how to get started. I haven't been actively developing C++ in the last few years.

erasmuswill, 2020-07-21:
A opening and closing state would be ideal but this has not been added to HASS as yet. :(

glmnet, 2020-07-21:
I see my covers shows opening or closing while they are doing so. Using time based cover. 

Are you talking about template cover or time based cover. We have both

erasmuswill, 2020-07-21:
@glmnet I'm talking about having a template cover which accepts "closed" or "open" state that estimates the position of the cover by using time.
e.g., My gate takes 30 seconds to open. The gate controller does not provide position to me, only action (opening, closing, open, closed or a couple of errors) so I'd want to estimate that, if 15 seconds have elapsed since the gate started opening, the position would be 50%. It would be nice to be able to open my gate half way by using the start command and then after 50% of the time, the stop command.

erasmuswill, 2020-07-21:
I see https://esphome.io/components/cover/template.html#cover-template-publish-action states that opening and closing states are supported (not sure what idle does) but the [lambda example](https://esphome.io/components/cover/template.html#template-cover) only defines open and closed.

glmnet, 2020-07-21:
If you go with the template cover you have to implement almost everything, i.e. the component is dumb and will not do anything by itself.

The endstop cover and timbased cover are components with logic. The https://esphome.io/components/cover/endstop.html should suit you better.

Why are not you considering it?

It lacks accepting feedback of actual cover state though.

erasmuswill, 2020-07-21:
> It lacks accepting feedback of actual cover state though.

This is the reason why not.

I've figured out how to set up the opening and closing and I think that should be enough to tide me over for the time being :)