esphome/feature-requests #817: support raspberry as building target

msmarks, 2020-07-22:
**Describe the problem you have/What new integration you would like**

Currently, ESPHome supports ESP32 and ESP8266 as building targets, and thanks to YAML syntax file, It's easy to make sensors and switch work by only a few lines. I am thinking if it is possible to support raspberry pi as a building target. YAML syntax file would make it easy to control raspberry pi's GPIO, SPI, I2C bus, and MQTT or native API would make it easy to integrate raspberry to home assistant.

**Please describe your use case for this integration and alternatives you've tried:**

I am doing some DIYs connecting raspberry to some sensors, cameras, buttons, screens. I have to deal with GPIOs, i2c, and MQTT. It is borning and makes me miss ESPHOME's  YAML syntax.

**Additional context**

glmnet, 2020-07-22:
Almost duplicate of #812

lwqcz, 2020-07-26:
I rather understand your reasons @msmarks but please @OttoWinter do not support this idea. It would possibly rather further thin scarce human resources so much needed elsewhere. 🙏