esphome/feature-requests #818: PCF8574 custom characters

RichardDern, 2020-07-23:
**Describe the problem you have/What new integration you would like**

I'd like to get an implementation of LiquidCrystal's method [createChar](https://www.arduino.cc/en/Reference/LiquidCrystalCreateChar).

**Please describe your use case for this integration and alternatives you've tried:**

I think it would be a nice addition to be able to define byte arrays for custom chars in display's definition. Maybe something like this:

```
- platform: lcd_pcf8574
  dimensions: 16x2
  address: 0x27
  custom_characters:
    - my_custom_char1: <character definition>
    - my_custom_char2: <...>
```

**Additional context**

Misiu, 2020-07-28:
@RichardDern I like the idea, but how will you call them?
They must be somehow callable/usable from lambda.

RichardDern, 2020-07-28:
The same as in LiquidCrystal's library (using ```byte(n)``` or an equivalent).

I guess in espHome we could display them this way:

```
it.printf(12, 0, "\x0"); // For the first defined custom char
```

nistvan86, 2020-11-28:
Custom characters would be very useful. They allow more economic use of the limited amount of display area.
Since it's only possible to define 8 characters at a time, it would be nice to have a set of characters and be able to switch between them between screen refreshes.

There's also big fonts implemented with this feature, which could be used to e.g.  [show a bigger clock](http://i0.wp.com/woodsgood.ca/projects/wp-content/uploads/BigFontDisplay_p2.jpg). Though this would be harder to implement, since this is essentially a virtual display on top of the display from the print APIs standpoint.

ESonya, 2021-06-03:
Hi, All!

I have made several changes to the library lcd_base. They allow the use of downloadable symbols. 
I have added the following features: 
printc(uint8_t column, uint8_t row, uint8_t str)
loadDeg(uint8_t location)
loadSmil(uint8_t location)
loadBel(uint8_t location)
createChar(uint8_t location, uint8_t charmap[])
[image]
[lcd_base.zip](https://github.com/esphome/feature-requests/files/6591786/lcd_base.zip)

ESonya, 2021-06-03:
For use, I replace two files during compilation in the esp_myproject\src\esphome\components\lcd_base directory.

Examples of using: 
```
display:
  - platform: lcd_pcf8574
    dimensions: 20x4
    address: 0x27
    lambda: |-
      it.strftime("%H:%M ", id(esptime).now());
      //it.createChar(6, 0x4,0xe,0xe,0xe,0x1f,0x0,0x4, 0x0);
      uint8_t const bell[8]  = {0x4,0xe,0xe,0xe,0x1f,0x0,0x4, 0};
      it.createChar(0, bell); // load custom character
      //it.loadBel(0);  // load predefined character
      it.printf(6, 0, "%.1f%c %.0f%c", id(hdc1080_temperature).state, 0x7, id(hdc1080_humidity).state, 1);
      it.printf(15, 1, "%c", char(round((id(vcc).state-3)/0.3)+2));  // print battery icon with current charge state
      it.printc(15, 0, 0);  //print character with code 0 (we load bell)
sensor:
  - platform: adc
    pin: VCC
    name: "VCC Voltage"
    id: vcc

```

nistvan86, 2021-06-03:
@ESonya your solution looks great :)
Have you noticed any slowdown with this? If I'm not mistaken this will upload the custom character to the LCD controller's memory on each refresh (since it's in the lambda). This has a nice bonus however that you can use the same character code for animated symbols and replace it's current bitmask during refresh. :) But it would be nice to have a way to only define new characters once at display initialization.

ESonya, 2021-06-03:
Yes, I know that in my version, createChar is loading the symbol every time display is refreshed.
Therefore, I added commonly used symbols to the display initialization routine.
How else to do it, I do not know.
```
static const uint8_t speaker[8]= {B00001,B00011,B01101,B01101,B01101,B00011,B00001,B00000};
static const uint8_t key[8]    = {B01110,B10001,B01110,B00100,B00100,B01100,B01100,B00000};
static const byte bat_empty[8] = {B00100,B11111,B10001,B10001,B10001,B10001,B11111,B00000};
static const byte bat_25[8]    = {B00100,B11111,B10001,B10001,B10001,B11111,B11111,B00000};
static const byte bat_50[8]    = {B00100,B11111,B10001,B10001,B11111,B11111,B11111,B00000};
static const byte bat_75[8]    = {B00100,B11111,B10001,B11111,B11111,B11111,B11111,B00000};
static const byte bat_full[8]  = {B00100,B11111,B11111,B11111,B11111,B11111,B11111,B00000};
static const uint8_t degree[8] = {B01100,B10010,B10010,B01100,B00000,B00000,B00000,B00000};

void LCDDisplay::setup() {
{ski[} 
  this->command_(LCD_DISPLAY_COMMAND_SET_CGRAM_ADDR);
    // load 8 predefined symbols: speaker, key, battery 5 types, degree icon
	for (int i=0; i<8; i++) {this->send(speaker[i], true);}
	for (int i=0; i<8; i++) {this->send(key[i], true);}
	for (int i=0; i<8; i++) {this->send(bat_empty[i], true);}
	for (int i=0; i<8; i++) {this->send(bat_25[i], true);}
	for (int i=0; i<8; i++) {this->send(bat_50[i], true);}
	for (int i=0; i<8; i++) {this->send(bat_75[i], true);}
	for (int i=0; i<8; i++) {this->send(bat_full[i], true);}
	for (int i=0; i<8; i++) {this->send(degree[i], true);}

```

nagyrobi, 2022-06-12:
See also https://github.com/esphome/feature-requests/issues/1766

ssieb, 2022-06-12:
You can print a null character with printf.  `printf("%c", 0);`