esphome/feature-requests #820: IR Remote Climate fujitsu_general

Enzokot, 2020-07-25:
**Describe the problem you have/What new integration you would like**

**Please describe your use case for this integration and alternatives you've tried:**

**Additional context**

glmnet, 2020-07-25:
This is already available `dev` version of ESPHome see here in https://next.esphome.io/components/climate/ir_climate.html 

It will be available in next major version, if you want to use this right now check FAQ about running different versions of ESPHome

Enzokot, 2020-07-26:
@glmnet feature request for this
[image]

Enzokot, 2021-03-14:
Add in PR [#1577](https://github.com/esphome/esphome/pull/1577)

closed 2021-03-14 by Enzokot

glmnet, 2021-03-14:
Yes I'll be updating the missing "yes" there in the docs