esphome/feature-requests #822: ESPHome WiFi repeater

averyanalex, 2020-07-26:
**Describe the problem you have/What new integration you would like**
I searched in issues and documentation, nothing about WiFi repeating. I think it will very useful to add repeater functionality.

**Please describe your use case for this integration and alternatives you've tried:**
Use cases: no need to buy repeaters in big houses. Realisation (only repeater, no IoT): [https://github.com/martin-ger/esp_wifi_repeater](https://github.com/martin-ger/esp_wifi_repeater) 

**Additional context**
Previous issue was auto closed, and I decided to don't edit template.

OttoWinter, 2020-07-26:
Yeah that's not really going to happen. We already have enough problems with WiFi on these ESPs, adding repeater (with full NAT etc) will only make that worse.

Plus the project you linked is built for a completely different framework (not Arduino framework base), a different Lwip core and so on. It's going to be near impossible to integrate those changes directly into ESPHome.

lwqcz, 2020-07-26:
There are much better solutions to act as a WiFi repeater. Search for Ubiquity Unifi, Amplifi or TPLInk stuff. Using a microcontroller as a repeater would be at least very slow, if ever possible.

I suggest closing this issue ... 📕

closed 2020-07-26 by OttoWinter