esphome/feature-requests #824: Switch Case Algorithm

Heiner1967, 2020-07-27:
**Describe the problem you have/What new integration you would like**
I like to code for a ESP a user menu, usualy this is done with a switch-case-loop which is not available via ESPHome, neither the variables which would allow to store a string.

**Please describe your use case for this integration and alternatives you've tried:**
I intend to build a watering device which switch on various tabs to rain te garden, bedie a fix program coded, I need ability to overwrite via some buttons on the device

**Additional context**
I know libraries could be added in principle, but here we have base functionality in C++, hence no idea on how to make this functionality available.

glmnet, 2020-07-27:
can you be more specific on what you need?

Heiner1967, 2020-09-09:
Hi, sorry for the long time it took to answer. I like to use the C++ command "switch-case-loop" as it it is common practice to use this command for coding a menu.
I like with one button to go up and down in such menu, and with another to select for such menu...

Alternativly a if-then comand could be used but that becomes quite complex.

alve89, 2021-12-13:
Since this issue is still open and I'm also at a point where a switch-case-statement would be (probably) more efficient than writing this in a lambda statement, I'd also like to ask for this feature request.

Now one has to do this:
```
switch( id(idOfElement) ) {
  case 0:
    // Do something
    break;
    
  case 1:
    // Do something
    break;

  case XYZ:
    // Do something
   break;

  default:
    // Do something
   break;
}
```

This could be easily done with something like this:
on_...:
   - switch
       - element: myElement
       - case 0:
           - switch.turn_on: xyz
       - case 1:
           - switch.turn_off: abc

douglasmauro, 2023-02-23:
I'll drop a vote in here too.  Here's what I *thought* might work:

```
      switch (d_gdoor_status)
      {
        case "open": 
          it.printf(x, y, id(font_icons), TextAlign::BOTTOM_LEFT, ICON_garage_open); 
          break;
        case "closed":
          it.printf(x, y, id(font_icons), TextAlign::BOTTOM_LEFT, ICON_garage_closed); 
          break;
        case "closing":
          it.printf(x, y, id(font_icons), TextAlign::BOTTOM_LEFT, ICON_garage_work); 
          break;
        case "opening":
          it.printf(x, y, id(font_icons), TextAlign::BOTTOM_LEFT, ICON_garage_work); 
          break;          
        default:
          it.printf(x, y, id(font_icons), TextAlign::BOTTOM_LEFT, ICON_alert); 
          break;  
      }
```

ssieb, 2023-02-23:
In C++ you can only use switch for integer values.
I really don't understand what this issue is about.  If you're in the display lambda, you can only use C++, so use the `switch` or an `if` tree.