esphome/feature-requests #828: Lambda-less counted loops and infinite loops

AlexDanault, 2020-07-29:
**Describe the problem you have/What new integration you would like**

A new YAML action to simplify counted and infinite loops in automations

**Please describe your use case for this integration and alternatives you've tried:**

I hope I'm not alone thinking this, but scripting (automations) could benefit from two more structures usable with 100% YAML and 0% lambdas : counted loops ( `for (int i = 0; i < 5; i++) { ... } ` ) and infitine loops ( `while(true) { ... }` ).

Both can be done currently using a While action but rely on lambdas and even variables. I have a script that blinks a light (it is in a script so it can be started/stopped depending on events) and is a bit wordy. 

```yaml
- while:
    condition:
      lambda: 'return 1;'
    then:
      - switch.turn_on: red
      - delay: 350ms
      - switch.turn_off: red
      - delay: 350ms
```

**Additional context**

Counted loop need even mode lambda. Both could be simplified with a new Loop action:

```yaml
- loop:
    count: infinite    #or any positive integer for counted loops, or a lambda that returns an integer or null for infinite
    then:
      - switch.turn_on: red
      - delay: 350ms
      - switch.turn_off: red
      - delay: 350ms
```

Would this be a nice addition to EspHome ?

nagyrobi, 2022-07-08:
Implemented in PR above.

closed 2022-07-08 by nagyrobi

AlexDanault, 2022-07-08:
Awesome ! Thanks !