esphome/feature-requests #830: Support for 4D Systems IOD-09

rthorntn, 2020-08-02:
**Describe the problem you have/What new integration you would like**

Please add support for 4D Systems [IOD-09](https://4dsystems.com.au/products/iot-display-modules/iod-09-display-modules/sk-iod-09) IoD

**Please describe your use case for this integration and alternatives you've tried:**

N/A

**Additional context**

It uses a ESP8266 and you can put [Arduino](https://github.com/4dsystems/GFX4DIoD9) on them, page 11 of the [datasheet](https://4dsystems.com.au/mwdownloads/download/link/id/201/) talks about the Arduino install.

rthorntn, 2020-08-04:
4D Systems confirmed it uses the ST7735S

nagyrobi, 2022-07-07:
Implemented: https://esphome.io/components/display/st7735.html