esphome/feature-requests #831: Support for Wio Terminal

rthorntn, 2020-08-02:
**Describe the problem you have/What new integration you would like**

Please add support for [WIO Terminal](https://www.seeedstudio.com/Wio-Terminal-p-4509.html) IoD

**Please describe your use case for this integration and alternatives you've tried:**

N/A

**Additional context**

It's a neat little device, low cost, 
2.4” LCD Screen
ARM Cortex-M4F (Microchip ATSAMD51P19)
4 MB External Flash, 192 KB RAM
Realtek RTL8720DN 2.4Ghz / 5Ghz Wi-Fi BLE / BLE 5.0
USB OTG Support
5-way switch, IMU, Microphone, Buzzer, microSD card slot, Light sensor & IR Emitter (940nm)
[Battery Chassis](https://www.seeedstudio.com/Wio-Terminal-Chassis-Battery-p-4516.html) add-on

OttoWinter, 2020-08-06:
> ARM Cortex-M4F (Microchip ATSAMD51P19)

Not an ESP, so not for *ESP*Home. Sorry

closed 2020-08-06 by OttoWinter

RoganDawes, 2021-08-05:
It may be worth revisiting this decision. The hardware architecture of the Wio Terminal aligns very closely with the work I am currently doing with the STM32 port, as well as extending the API to communicate over other transports than just AsyncTCP. i.e. API over Serial.
It appears that there is an Arduino environment for the RTL8720 (under which the Seeed erpcWifi library was built), so there are a couple of options:
- Build ESPHome for RTL8720, as well as ESPHome for SAMD51, and use my API over Stream work to allow them to communicate.
- Build ESPHome for the SAMD51, and use the erpcWiFi library to allow the SAMD51 to access WiFi functionality directly.