esphome/feature-requests #833: [Request] Add support for Platform IO Remote agents

evil-dog, 2020-08-03:
**Describe the problem you have/What new integration you would like**

I run ESPHome in docker on my server in a closet. This makes it difficult to do initial flashing of new devices. Currently it appears that esphome only supports serial flashing of new devices. OTA flashing is only available for devices that have already had esphome flashed.

PlatformIO supports connecting to remote agents and flashing devices connected to those remote agents. Would be nice if I could do this from the ESPHome UI. 

**Please describe your use case for this integration and alternatives you've tried:**

Makes initial flashing of new devices with esphome firmware.

I've tried using esptool, but this can be overly complicated for esp32 devices.

I've used the platformio remote agent to do the initial flashing, but it requires me to ssh to my server and open a cli into the docker container and run the pio command manually.

**Additional context**

https://docs.platformio.org/en/latest/plus/pio-remote.html

OttoWinter, 2020-08-06:
> Currently it appears that esphome only supports serial flashing of new devices.

Same limitation applies to platformio remote (the OTA there uses the arduino OTA feature, which first has to be flashed to the ESP too).

I'm not quite sure on the use-case either. If you're going to install platformio remote on a server-like machine, why not directly install ESPHome on that machine then? Then you could just use a REST command (or any other mechanism) to trigger the build+upload.

evil-dog, 2020-08-06:
The use case is laziness. I want to run esphome on my server that is always on, but I don't want to have to go to the garage every time I want to flash a new device for the first time to plug it into one of the server's USB ports.

I could (and do) run the platformio remote agent on my desktop (or a rpi) and connect the new device to it.

Currently I have to ssh to my server, access the cli in the docker running esphome, then run the platformio remote command to upload using the remote agent.

The request here is to avoid doing the ssh>docker>pio remote cli stuff. It would be nice to be able to select pio remote in the OTA menu, or even better would be to have a list of the remote agents in the OTA menu since you could have more than one.

alextrical, 2024-10-21:
> I'm not quite sure on the use-case

Im working as part of a HW & SW development team, and shipments of hardware don't arrive at all members of the teams desks at the same time. Having remote access for the software development team would have solved a lot of headaches we have faced over the last 3 months.

We are also looking for this feature to allow for CI CD pipelines inside of GitHub, to confirm that any changes made are compatible with real world hardware, before giving code to an end user that could potentially cause issues on their setup.

This feature would be beneficial for team development and testing of ESPHome, rather than an individuals home lab setup.

Quote from Platformio website:

[Use Cases](https://docs.platformio.org/en/latest/plus/pio-remote.html#use-cases)
**Remote Unit Testing:**

Instruct any of [Continuous Integration](https://docs.platformio.org/en/latest/integration/ci/index.html#ci) services to run remote tests on a physical device. See the documentation for [Remote Test Runner](https://docs.platformio.org/en/latest/advanced/unit-testing/runner.html#unit-testing-runner-remote).

How does it work?

You commit new changes to your source code repository

[Continuous Integration](https://docs.platformio.org/en/latest/integration/ci/index.html#ci) service deploys unit tests to a remote agent

[Unit Testing](https://docs.platformio.org/en/latest/advanced/unit-testing/index.html#unit-testing) engine runs tests on a physical device, process them, and send results

[Continuous Integration](https://docs.platformio.org/en/latest/integration/ci/index.html#ci) service prints results in human readable format

If one test fails, current CI build will fail too.

**Board Farm:**

A similar concept as described in “Remote Unit Testing” above. Let’s imagine that you need to test some logic on the unlimited number of target devices. Very often it can be the same hardware prototype but with different factory revisions.

You connect these devices via USB hub to PC and instruct PlatformIO Remote Development Solution to process your test on ALL targets connected to a specific agent. See documentation below.

**Remote Serial Monitor:**

Sometimes you don’t have physical access to a target device but you need to read data from some serial port. PlatformIO Remote Development Solution allows you to connect to a remote agent and list connected devices with their serial ports. See [pio remote device monitor](https://docs.platformio.org/en/latest/core/userguide/remote/cmd_device.html#cmd-remote-device-monitor) command for details.

**Cloud IDE:**

Program your devices from anywhere in the world using the most popular [Cloud IDEs](https://docs.platformio.org/en/latest/integration/ide/index.html#ide-cloud). You do not need to install any extra software, no need to have static IP or open network ports. Everything works out of the box.

**Devices behind card sized PC:**

Work with your favorite development environment and program devices connected to card-sized PC (Raspberry Pi, Cubie Board, etc.). You do not need to open SSH ports, install any extra Linux packages, toolchains.
```