esphome/feature-requests #834: [Request] Refactor UI Dashboard

evil-dog, 2020-08-03:
**Describe the problem you have/What new integration you would like**

I have quite a few devices that are 99% the same and basically the only difference is the name of the device.

It would be nice if the UI would let you create a "device class" that has the main part of the config (e.g. everything you would typically configure today in the device).  But then have the ability to create instantiations of that device class and customise certain config settings (i.e. the name and some other parameters that might be specific to individual devices). 

I know there is a way to kind of do this today, but IIUC it requires creating files outside of the UI and then importing them. This is rather clunky and not user friendly for more novice users.

**Please describe your use case for this integration and alternatives you've tried:**

If I buy 10 of the exact same model LED light bulbs, I have to create a full 10 different devices in the current UI.

With the this requested feature, I could create 1 overall class for that specific LED light bulb model, then have 10 sub device instantiations that setup each device implementation with the proper name, ip, etc...

I've tried using the substitutions component, but couldn't seem to get it to work.

**Additional context**

glmnet, 2020-08-04:
Substitution should do what you describe here. If you need more advanced stuff see inclusions and packages which are quite powerful (available in beta). 

I really don't know how dashboard will handle them, but that is cosmetic and should just work

evil-dog, 2020-08-06:
So I think this would be a UI enhancement to better organize devices and better support substitutions.

AFAIK, the only way to do substitutions in the UI is to create a device with the substitutions defined. This device would always be there and never show as connected.  You then create your actual devices and import the yml for the base device, these devices would show up as live.

There is also no linking between the base and the real devices, so if you update the code in the base device there is no way for the UI to know to indicate the other devices that need to be recompiled and uploaded.

Packages look like a nice way to create repeatable code segments, like wifi settings, but I'm not sure it would be the same as what I'm talking about here. I'm sure you could use it in a similar way, but it'll probably come down to how it's implemented in the UI.

glmnet, 2020-08-06:
> the only way to do substitutions in the UI

More technically substitutions works on the same file, but that is not a limitation.

> if you update the code in the base device there is no way for the UI to know to indicate the other devices that need to be recompiled

That is not available even if you don't include any file, in fact, some people are requesting a solution for this, this is definitely another feature request

The way it should work, IMHO is to have the feature be supported on yaml / commands only first, then dashboard can support the feature on top of that.

We could support your first feature by, e.g. how secrets are implemented, where the dashboard doesn't list list secrets.yaml as a node, we could have other files not listed as nodes too, that will look way more of a specific request, would that suit you?, or if you have other specific ideas that will help.

torwag, 2020-08-15:
Wouldn't it be the easiest to enable folders and a recursive scan for those. People could still hide files and folder using the .name-notation (like on the file system). But could start to organize their config files in any way they want (room-wise, device-wise, core units, etc. That would also become more relevant with the new package feature where people might start to split config files even further.