esphome/feature-requests #835: OV7670 Support

ShonP40, 2020-08-04:
**Describe the problem you have/What new integration you would like**

I have an OV7670 camera and when trying to set it up using the ESP32_camera module, I get an error message that says that the camera isn't supported.

**Please describe your use case for this integration and alternatives you've tried:**

I have an OV7670 camera connected to an ESP32 and used [this](https://github.com/bitluni/ESP32CameraI2S) before.

**Additional context**

I found [this](https://github.com/espressif/esp32-camera/pull/88) pull request about adding a driver for that camera on the ESP-Camera library, but its been inactive for a while now.

OttoWinter, 2020-08-08:
Yes until https://github.com/espressif/esp32-camera/pull/88 is merged (and included in the base framework we use), this is blocked.

ShonP40, 2020-10-14:
Hey,
They merged the pull request yesterday.
Is it suppose to work now, or will I need to wait for an ESPHome update?

ShonP40, 2020-10-14:
Its still giving me the same error: `camera_probe(): Detected camera not supported.`
I guess that I'll need to wait for an ESPHome update

ShonP40, 2020-10-23:
Disappointed that this wasn't added with the latest update...

ShonP40, 2021-01-06:
Closing this issue because I am no longer using this camera and because I am getting ignored anyways.

closed 2021-01-06 by ShonP40