esphome/feature-requests #836: Add the STM VL53L1X sensor (4m version of the VL53L0X)

torwag, 2020-08-05:
**Describe the problem you have/What new integration you would like**
I would like to see support for the STM VL53L1X sensor for ranging distances and detecting objects.

**Please describe your use case for this integration and alternatives you've tried:**
I looked into the integration of the STM VL53L0X in the hope that it would be very similar or even fit. However, looking into the specs it does not unfortunately,  there is a complete different set of registers. 

**Additional context**
Polou which afaik is the base for the integration in esphome, offers two distinct projects to integrate both versions.  
[vl53l0x](https://github.com/pololu/vl53l0x-arduino)
[vl53l1x](https://github.com/pololu/vl53l1x-arduino)
Albeit I do not know about the state of the vl53l1x project, as I did not yet test it.
Unfortunately, even it is said that the esphome integration is based on the pololu/vl53l0x-arduino library, I could not see enough similarities to get into it and to see how something similar could be done with the vl53l1x project.
As of now I could not find the specific API description of STM either, I found some C-lib but not yet the I2C communication protocol. A link would be highly appreciated.

rspaargaren, 2020-08-07:
As far as I can read there is no difference in controlling 1x and 0x. So I the same library can be used. There is now also option to set 0x in long range mode in esp. For the 1x there are 3 profiles but short medium and long range. But I don't know what is default when u use 0x code.

https://wolles-elektronikkiste.de/vl53l0x-und-vl53l1x-tof-abstandssensoren

torwag, 2020-08-23:
I don't think it is the same. The Pololu libs look quite different. Albeit on the discord channel was a post of someone who created a custom component based on the pololu libs for the vl53l1x. Will test this and if it works. It might be not to hard to make this a contribution to main.
Reference:
[](https://github.com/jardous/tof_vl53l1x)

torwag, 2020-08-30:
The implementation of this sensor is a bigger task as it uses 16 bit address registers (whereas the vl53l0x uses 8 bits). As the current implementation of the I2C bus in esphome uses 8bit address registers as standard, it takes more effort to get this unit working. See #875

osos, 2022-09-17:
I made a simple setup with small modifications to https://github.com/jardous/tof_vl53l1x.

esphome yaml:
```
substitutions:
  devicename: distancetest

esphome:
  name: '${devicename}'
  platform: ESP32
  #board: lolin32
  board: mhetesp32minikit
  # https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html
  includes:
    - tof_vl53l1x.h
  libraries:
    - "Wire"
    - "VL53L1x"

wifi:
...
captive_portal:

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:

ota:

i2c:
  sda: 21
  scl: 22
  scan: false
  frequency: 400kHz

sensor:
- platform: custom
  lambda: |-
    auto my_VL53L1X_sensor = new VL53L1XCustomSensor();
    my_VL53L1X_sensor->set_update_interval(2000); // define update interval
    App.register_component(my_VL53L1X_sensor);
    return {my_VL53L1X_sensor};
  sensors:
    name: "Distance"
    accuracy_decimals: 0
    unit_of_measurement: "mm"
```

Placed the tof_vl53l1x.h in same directory as the yaml file. with this content:

```
#include "esphome.h"

#include <Wire.h>
#include <VL53L1X.h>

class VL53L1XCustomSensor : public PollingComponent, public Sensor {

 private:
  VL53L1X tof_sensor;

 public:
  // constructor
  VL53L1XCustomSensor() : PollingComponent(15000) {} // polling every 15s

  void setup() override {
    // This will be called by App.setup()
    Wire.begin();
    Wire.setClock(400000); // use 400 kHz I2C

    tof_sensor.setTimeout(500);
    tof_sensor.setAddress(0x29);
    if (!tof_sensor.init()) {
      ESP_LOGE("VL53L1X custom sensor", "Failed to detect and initialize sensor!");
      return;
    }

    tof_sensor.setDistanceMode(VL53L1X::Long);
    tof_sensor.setMeasurementTimingBudget(250000);	// 1000us = 1ms

    ESP_LOGI("VL53L1X custom sensor", "initialised");

    //tof_sensor.startContinuous(250);	// ms
  }

  void update() override {
    //uint16_t distance_mm = tof_sensor.read(false);
    uint16_t distance_mm = tof_sensor.readSingle();
    
    if (!tof_sensor.timeoutOccurred()) {
      publish_state(distance_mm);
    } else {
      ESP_LOGE("VL53L1X custom sensor", "Timeout during read().");
    }
  }
};
```

jcamer, 2022-09-30:
> I made a simple setup with small modifications to https://github.com/jardous/tof_vl53l1x.
> 
> esphome yaml:
> 
> ```
> substitutions:
>   devicename: distancetest
> 
> esphome:
>   name: '${devicename}'
>   platform: ESP32
>   #board: lolin32
>   board: mhetesp32minikit
>   # https://doc.riot-os.org/group__boards__esp32__mh-et-live-minikit.html
>   includes:
>     - tof_vl53l1x.h
>   libraries:
>     - "Wire"
>     - "VL53L1x"
> 
> wifi:
> ...
> captive_portal:
> 
> # Enable logging
> logger:
>   level: DEBUG
> 
> # Enable Home Assistant API
> api:
> 
> ota:
> 
> i2c:
>   sda: 21
>   scl: 22
>   scan: false
>   frequency: 400kHz
> 
> sensor:
> - platform: custom
>   lambda: |-
>     auto my_VL53L1X_sensor = new VL53L1XCustomSensor();
>     my_VL53L1X_sensor->set_update_interval(2000); // define update interval
>     App.register_component(my_VL53L1X_sensor);
>     return {my_VL53L1X_sensor};
>   sensors:
>     name: "Distance"
>     accuracy_decimals: 0
>     unit_of_measurement: "mm"
> ```
> 
> Placed the tof_vl53l1x.h in same directory as the yaml file. with this slightly modified content:
> 
> ```
> #include "esphome.h"
> 
> #include <Wire.h>
> #include <VL53L1X.h>
> 
> 
> class VL53L1XCustomSensor : public PollingComponent, public Sensor {
> 
>  private:
>   VL53L1X tof_sensor;
> 
>  public:
>   // constructor
>   VL53L1XCustomSensor() : PollingComponent(15000) {} // polling every 15s
> 
>   void setup() override {
>     // This will be called by App.setup()
>     Wire.begin();
>     Wire.setClock(400000); // use 400 kHz I2C
> 
>     tof_sensor.setTimeout(500);
>     tof_sensor.setAddress(0x29);
>     if (!tof_sensor.init()) {
>       ESP_LOGE("VL53L1X custom sensor", "Failed to detect and initialize sensor!");
>       return;
>     }
> 
>     tof_sensor.setDistanceMode(VL53L1X::Long);
>     tof_sensor.setMeasurementTimingBudget(250000);	// 1000us = 1ms
> 
>     ESP_LOGI("VL53L1X custom sensor", "initialised");
> 
>     //tof_sensor.startContinuous(250);	// ms
>   }
> 
>   void update() override {
>     //uint16_t distance_mm = tof_sensor.read(false);
>     uint16_t distance_mm = tof_sensor.readSingle();
>     
>     if (!tof_sensor.timeoutOccurred()) {
>       publish_state(distance_mm);
>     } else {
>       ESP_LOGE("VL53L1X custom sensor", "Timeout during read().");
>     }
>   }
> };
> ```

I have tried the steps you have, but when I go to install I am getting the error below:

In file included from src/tof_vl53l1x.h:4:0,
                 from src/main.cpp:23:
src/VL53L1X.h: In member function 'virtual float MyCustomSensor::get_setup_priority() const':
src/VL53L1X.h:6:54: error: 'XXXX' is not a member of 'esphome::setup_priority'
   float get_setup_priority() const override { return esphome::setup_priority::XXXX; }
                                                      ^
In file included from src/main.cpp:23:0:
src/tof_vl53l1x.h: At global scope:
src/tof_vl53l1x.h:10:3: error: 'VL53L1X' does not name a type
   VL53L1X tof_sensor;
   ^
src/tof_vl53l1x.h: In member function 'virtual void VL53L1XCustomSensor::setup()':
src/tof_vl53l1x.h:21:5: error: 'tof_sensor' was not declared in this scope
     tof_sensor.setTimeout(500);
     ^
src/tof_vl53l1x.h:28:32: error: 'VL53L1X' has not been declared
     tof_sensor.setDistanceMode(VL53L1X::Long);
                                ^
src/tof_vl53l1x.h: In member function 'virtual void VL53L1XCustomSensor::update()':
src/tof_vl53l1x.h:38:28: error: 'tof_sensor' was not declared in this scope
     uint16_t distance_mm = tof_sensor.readSingle();_
====================

What does your VL53L1X.h file look like?

thanks,

John

stanogustafik, 2024-06-26:
I tried this solution, but it just doesn´t work... please would you consider to build platform for these devices? there are no other sensors detecting distance to 400cm... this is from my logs after trying the solution by @osos :

`INFO ESPHome 2024.6.3
INFO Reading configuration /config/esphome/my-sensor.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing my-sensor (board: esp32dev; framework: arduino; platform: platformio/espressif32@5.4.0)
--------------------------------------------------------------------------------
Removing unused dependencies...
Library Manager: Removing VL53L1X @ 1.3.1
INFO Removing VL53L1X @ 1.3.1
Library Manager: VL53L1X@1.3.1 has been removed!
INFO VL53L1X@1.3.1 has been removed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Library Manager: Installing VL53L1x
Library Manager: Warning! More than one package has been found by VL53L1x requirements:
Library Manager:  - pololu/VL53L1X@1.3.1
Library Manager:  - mbed-jvfausto/VL53L1X@0.0.0+sha.621552ff1de9
Library Manager:  - mbed-st-expansion-sw-team/VL53L1X@0.0.0+sha.f16727052990
Library Manager:  - mbed-1576028839/VL53L1X@0.0.0+sha.a71614c1b15b
Library Manager: Please specify detailed REQUIREMENTS using package owner and version (shown above) to avoid name conflicts
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Library Manager: VL53L1X@1.3.1 has been installed!
Dependency Graph
|-- AsyncTCP-esphome @ 2.1.3
|-- Wire @ 2.0.0
|-- VL53L1X @ 1.3.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.2.2
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
Compiling .pioenvs/my-sensor/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/custom/sensor/custom_sensor.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/esp32/gpio.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/esp32/preferences.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/esphome/ota/ota_esphome.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/i2c/i2c.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/i2c/i2c_bus_arduino.cpp.o
In file included from src/esphome/components/i2c/i2c_bus_arduino.cpp:3:
src/esphome/components/i2c/i2c_bus_arduino.h:38:3: error: 'TwoWire' does not name a type; did you mean 'TwoWire_h'?
   TwoWire *wire_;
   ^~~~~~~
   TwoWire_h
Compiling .pioenvs/my-sensor/src/esphome/components/i2c/i2c_bus_esp_idf.cpp.o
Compiling .pioenvs/my-sensor/src/esphome/components/json/json_util.cpp.o
src/esphome/components/i2c/i2c_bus_arduino.cpp: In member function 'virtual void esphome::i2c::ArduinoI2CBus::setup()':
src/esphome/components/i2c/i2c_bus_arduino.cpp:21:5: error: 'wire_' was not declared in this scope
     wire_ = &Wire;
     ^~~~~
src/esphome/components/i2c/i2c_bus_arduino.cpp:21:5: note: suggested alternative: 'writev'
     wire_ = &Wire;
     ^~~~~
     writev
src/esphome/components/i2c/i2c_bus_arduino.cpp:21:14: error: 'Wire' was not declared in this scope
     wire_ = &Wire;
[... 19 lines cut ...]
   ^~~~~
src/esphome/components/i2c/i2c_bus_arduino.cpp:53:3: note: suggested alternative: 'writev'
   wire_->begin(static_cast<int>(sda_pin_), static_cast<int>(scl_pin_));
   ^~~~~
   writev
src/esphome/components/i2c/i2c_bus_arduino.cpp: In member function 'virtual esphome::i2c::ErrorCode esphome::i2c::ArduinoI2CBus::readv(uint8_t, esphome::i2c::ReadBuffer*, size_t)':
src/esphome/components/i2c/i2c_bus_arduino.cpp:125:16: error: 'wire_' was not declared in this scope
   size_t ret = wire_->requestFrom((int) address, (int) to_request, 1);
                ^~~~~
src/esphome/components/i2c/i2c_bus_arduino.cpp:125:16: note: suggested alternative: 'writev'
   size_t ret = wire_->requestFrom((int) address, (int) to_request, 1);
                ^~~~~
                writev
src/esphome/components/i2c/i2c_bus_arduino.cpp: In member function 'virtual esphome::i2c::ErrorCode esphome::i2c::ArduinoI2CBus::writev(uint8_t, esphome::i2c::WriteBuffer*, size_t, bool)':
src/esphome/components/i2c/i2c_bus_arduino.cpp:179:3: error: 'wire_' was not declared in this scope
   wire_->beginTransmission(address);
   ^~~~~
src/esphome/components/i2c/i2c_bus_arduino.cpp:179:3: note: suggested alternative: 'writev'
   wire_->beginTransmission(address);
   ^~~~~
   writev
*** [.pioenvs/my-sensor/src/esphome/components/i2c/i2c_bus_arduino.cpp.o] Error 1
========================= [FAILED] Took 66.03 seconds =========================`

thank you

frederickjh, 2024-07-05:
The abstraction for reading and writing to 16 bit I2C registers was added to esphome/esphome#4844 and merged. #875 was closed as a result of this. Hopefully this paves the way  for support for the VL53L1x to be added.

baerrs, 2024-07-16:
I would love to see this added as well.

soldierkam, 2024-08-09:
My WIP version of the component: https://github.com/soldierkam/vl53l1x_sensor

clomads, 2024-08-28:
@soldierkam just loaded it on one of my ESP32_C3 dev boards and it's working great at 100ms update_interval in SHORT mode.

glennlawyer, 2024-09-07:
I would also like to see the VL53L1X added to esphome; I'm working with the M5 Stack "Unit-ToF4M" which uses the VL53L1CX chip.

dersch81, 2024-09-12:
Me too. I also want to use two VL53L1X at the same ESPhome device.

bojanpotocnik, 2024-11-05:
Seems like @soldierkam (https://github.com/soldierkam/vl53l1x_sensor) and @mrtoy-me (https://github.com/mrtoy-me/esphome-my-components/tree/main/components/vl53l1x, src https://github.com/esphome/feature-requests/issues/1697#issuecomment-1607510945) have working implementations, are there any plans of opening a PR to eventually merge this?

archfolf, 2024-12-06:
I got @soldierkam (https://github.com/soldierkam/vl53l1x_sensor) to work, but it won't compile with esp-idf framework

soldierkam, 2025-02-02:
> I got [@soldierkam](https://github.com/soldierkam) (https://github.com/soldierkam/vl53l1x_sensor) to work, but it won't compile with esp-idf framework

@archfolf What was the problem? I checked on ESP32C3 and it compiles with esp-idf. It would be great if you could submit an issue on the project page.

mrtoy-me, 2025-02-03:
After a lot of testing... I have created a new esphome external component for VL53L1X based on Polulo Arduino library. It located at:

https://github.com/mrtoy-me/esphome-components-test

I think this implementation is better(more stable) than the previous implementation I did - mentioned above by @bojanpotocnik.

I will eventually delete the other version, as this is much preferred.

andywarburton, 2025-02-19:
@mrtoy-me thanks for doing this - the latest update stopped my custom sensor from working and I was just starting to look into how to create an external component myself. It would be great if this could be folded into ESPHome officially some day!

andywarburton, 2025-02-20:
I have just installed and tested this and can confirm it works great!

B48D81EFCC, 2025-03-07:
Thanks all,
I did not pay attention and bought some  VL53L1X
Especially thanks to @mrtoy-me your external component works great.

Maybe this can be merged into the official code base.

mrtoy-me, 2025-09-29:
I have deprecated and moved to the following. There are some issues raised with the newer esphome versions which I will look at soon

https://github.com/mrtoy-me/esphome-vl53l1x