esphome/feature-requests #838: Summation module

bseishen, 2020-08-05:
Summation module that works like an integrator without floating point math and hard sampling rates.

**Please describe your use case for this integration and alternatives you've tried:**
Use Case:
Sensors that are really slow where every pulse counts. Rain gauges, etc. That are measured on pulses per hour/pulses per day.

A pulse counter fed into a left aligned integrator set at the same time interval should give the same result, but will run into compounding floating point errors eventually. Also doesn't allow for flexibility in the sampled time interval. I.e. 15min.

**Additional context**

Also would Nyquist sampling theory be an issue here too?

glmnet, 2020-08-05:
Is this a duplicate of https://github.com/esphome/feature-requests/issues/763?

It has a PR already:

https://github.com/esphome/esphome/pull/1186

bseishen, 2020-08-05:
Closing, duplicate feature request. Thanks! Fingers crossed for merge of 1186.

closed 2020-08-05 by bseishen