esphome/feature-requests #841: Recieve BLE messages on ESP32

reesericci, 2020-08-06:
**Describe the problem you have/What new integration you would like**
I would really like it if it was possible to receive messages from BLE devices.

**Please describe your use case for this integration and alternatives you've tried:**
I would be able to have a BLE device send the message "open" to trigger a cover to open and "close" to close a cover

**Additional context**
I would manually program these messages to be sent from a device with something like CircuitPython or Arduino. 
Wouldn't it be using UART over bluetooth?

buxtronix, 2020-08-07:
BLE integration is on its way, a PR is already out:

https://github.com/esphome/esphome/pull/1177

Also, if the cover you're thinking about is the AM43 devices, it will follow the above PR.

reesericci, 2020-08-07:
No, I'm not I'm more thinking of setting up an Adafruit Circuit Playground Bluefruit on circuitpython to send messages. I'll go take a look at that PR though.

reesericci, 2020-08-08:
ok

reesericci, 2020-08-08:
I'm using a ESP32 connected to a gate remote with a US sensor, and want to send BLE messages to it to open and shut the gate.

nagyrobi, 2022-07-06:
Implemented via https://esphome.io/components/esp32_ble_tracker.html

closed 2022-07-06 by nagyrobi