esphome/feature-requests #842: Serve web ui CSS/JS from dashboard

Almighty-Alpaca, 2020-08-08:
**Describe the problem you have/What new integration you would like**
I want to use the ESPHome Dashboard running on my Raspberry Pi to serve the CSS and JS files for my ESPHome Web UIs.

**Please describe your use case for this integration and alternatives you've tried:**

I'd like to use ESPHome completely locally even when my internet connection has issues. Without running my own local webserver to host the CSS/JS this currently isn't possible. Since many users probably have the dashboard running and accessible via a domain or static IP anyway it would be very convenient to use be able to use it to host those files. This solution also has the benefit of keeping the files up-to-date as long as you update your ESPHome installation regularly (assuming they will ever change).

**Additional context**

glmnet, 2020-08-09:
In that case you'll need to setup the addon so it does allow access to http directly without ingress?

Btw currently you can host the css and js in the esp itself, that is supported in beta version right now. 

Yet another option will be hosting in Lovelace?

Almighty-Alpaca, 2020-08-11:
>In that case you'll need to setup the addon so it does allow access to http directly without ingress?

Well, the dashboard is accessible via HTTP in my local network anyway and all static files from the dashboard are accessible without being authenticated, so I don't think this is really an issue.

>Btw currently you can host the css and js in the esp itself, that is supported in beta version right now.

Interesting, didn't think that would be implemented considering #237 (although I have to admit that's a pretty old issue).

> Yet another option will be hosting in Lovelace?

I guess you're right, I didn't think about that option. I'm probably gonna do that for now. I still think just serving them from the dashboard would be a nice feature as it would require less configuration on the user's side.