esphome/feature-requests #844: ESP32 Camera Color Picker

hansjochensen, 2020-08-10:
**Describe the problem you have/What new integration you would like**

I want to read the single RGB or the luminance values of a specified pixel of an ESP32 camera picture directly within ESPhome, similar to a **Color Picker**. Provide a pixel coordinate (x,y) and get one of the RGB values in return. 
**Please describe your use case for this integration and alternatives you've tried:**

I  read my 3-phase Ferres power meter by detecting the red section of the revolving disk inside. My trials with luminance and RGB boards were not successful. Instead I installed an ESP32 Camera to detect the red stripe. It works fine, but is cumbersome and not suitable for higher power consumption. My current approach inside Home Assistant: Crop the camera image at the ROI to 1 pixel and save it every second as a jpg file. Read the byte containing the DC-part of the FFT luminance encoding with a commandline sensor using tail and hexdump. The sensor value varies sufficiently enough to detect a complete resolution of the disk, which is in my case equivalent to the consumption of 8 Watt.
**Additional context**

Such a feature would allow an ESP32 camera to act as a **generic sensor** in other application domains too.

hansjochensen, 2020-08-10:
Shown are the original image and three cop results. The red pixel is detected and increases the revolutions counter.
[image]

JeeCee1, 2020-08-10:
Hi Hans, what an awesome solution! Os your code online?